pikvm/pikvm #430: reset hdmi csi-2

pete1019, 2021-08-23:
Hi,

is there a way to reset the hdmi csi-2? Best would be from WebGui.

Why i am asking: on intel NUC on reboot it often happens that the signal is destroyed when entering Windows. I then need to unplug the HDMI cable and plug it back in. This would be difficult when using pikvm remote.

Thanks a lot!

mdevaev, 2021-08-23:
Try this:
```
v4l2-ctl --device=/dev/kvmd-video --clear-edid
v4l2-ctl --device=/dev/kvmd-video --set-edid=file=/etc/kvmd/tc358743-edid.hex --fix-edid-checksums
```

pete1019, 2021-08-23:
Thanks for the reply.

i tested it while it was still working and after the two commands i looks like the error can be replicated with those commands:
[image]

Tried it again right away, does not fix it. Need to unplug the cable and plug it again.

mdevaev, 2021-08-23:
If you don't use these commands, what does signal loss look like in windows?

pete1019, 2021-08-24:
> If you don't use these commands, what does signal loss look like in windows?

Same as you see above. The distortion is moving.
So the same when i use the second command. First command the picture is gone. Second command the picture looks like above, pixels moving.