pikvm/pikvm #431: Web Terminal doesn’t work on iPad

svh1985, 2021-08-23:
**Describe the bug**
I'm running the iPadOS 15 beta and I noticed the that the web terminal doesn't work. Not sure if it also applies for iPadOS 14.

**To Reproduce**
Open pikvm on an iPad running iPadOS 15.

**Expected behavior**
A working web terminal :)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: iPadOS 15
 - Browser [safari]

**Pi-KVM info:**
  - Raspberry Pi board version: 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: USB-HDMI
  - KVMD version: 

```
kvmd 3.18-1
kvmd-fan 0.1-1
kvmd-oled 0.3-1
kvmd-platform-v2-hdmiusb-rpi4 3.18-1
kvmd-webterm 0.40-1
```

  - uStreamer version: `ustreamer 4.5-1 `
  - Linux kernel: `Linux pikvm 5.10.52-9-ARCH #1 SMP Thu Aug 12 13:42:42 MSK 2021 armv7l GNU/Linux `

arch1mede, 2021-09-05:
Using iOS 14, unable to repo after disabling https

[image]

This shows us you have https enabled OR the ssl cert you are using is no longer valid.

svh1985, 2021-09-05:
@arch1mede its the default pikvm self generated certificate. Not sure if this is to be expected, if so, an option would be to switch to http. But I couldn't find it in the 😅. But if possible would rather use https of course.

arch1mede, 2021-09-05:
https://github.com/pikvm/pikvm/blob/master/pages/faq.md#i-cant-use-this-on-ios-the-web-ui-network-indicator-flashes-yellow

This is a known issue.

For me personally, I see no value in using https within a home network so I don't bother with getting a proper ssl. Also for me personally, there is no value in https even within a corp network. If someone wants to sniff the network on me boringly fixing a system, let them die of boredom.

svh1985, 2021-09-05:
Changed it according to the doc, rebooted, but now I get this: [image]

arch1mede, 2021-09-05:
Might be 2 things, try the IP or it might be that its the beta.

svh1985, 2021-09-06:
@arch1mede tried both IP and checked on my Mac it but also has the same issue.

Chrome console (macOS):
`::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)`

Safari console (macOS):
`Failed to load resource: The network connection was lost.`

my `/etc/kvmd/nginx/nginx.conf` config:
```
worker_processes 4;

# error_log /tmp/kvmd-nginx.error.log;
error_log stderr;

include /usr/share/kvmd/extras/*/nginx.ctx-main.conf;

events {
        worker_connections 1024;
        use epoll;
        multi_accept on;
}

http {
        types_hash_max_size 4096;
        server_names_hash_bucket_size 128;

        access_log off;

        include /etc/kvmd/nginx/mime-types.conf;
        default_type application/octet-stream;
        charset utf-8;

        sendfile on;
        tcp_nodelay on;
        tcp_nopush on;
        keepalive_timeout 10;
        client_max_body_size 4k;

        client_body_temp_path   /tmp/kvmd-nginx.client_body_temp;
        fastcgi_temp_path               /tmp/kvmd-nginx.fastcgi_temp;
        proxy_temp_path                 /tmp/kvmd-nginx.proxy_temp;
        scgi_temp_path                  /tmp/kvmd-nginx.scgi_temp;
        uwsgi_temp_path                 /tmp/kvmd-nginx.uwsgi_temp;

        include /etc/kvmd/nginx/kvmd.ctx-http.conf;
        include /usr/share/kvmd/extras/*/nginx.ctx-http.conf;

#       server {
#               listen 80;
#               listen [::]:80;
#               server_name localhost;
#               return 301 https://$host$request_uri;
#       }

        server {
#               listen 443 ssl http2;
#               listen [::]:443 ssl http2;
                server_name localhost;
#               include /etc/kvmd/nginx/ssl.conf;

                include /etc/kvmd/nginx/kvmd.ctx-server.conf;
                include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
        }
}
```

arch1mede, 2021-09-06:
iOS and OSX are 2 different things. Looks like your config is wrong. Should look like the following:

```
        server {
                listen 80;
                listen [::]:80;
                server_name localhost;
#               return 301 https://$host$request_uri;
#       }

#       server {
#               listen 443 ssl http2;
#               listen [::]:443 ssl http2;
#               server_name localhost;
#               include /etc/kvmd/nginx/ssl.conf;

                include /etc/kvmd/nginx/kvmd.ctx-server.conf;
                include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
        }
}
```

svh1985, 2021-09-07:
The changes* you shared fixes the issue on macOS, on iPadOS and iOS it "reverted" back to the initial issue. Might be iOS 15 related; I noticed that the Unraid web terminal, which I think uses the same webterminal, also doesn't work.

*The changes are bit different from the Community FAQ article you mentioned earlier btw.

Thanks!

arch1mede, 2021-09-07:
FYI that verbiage was from the FAQ which is an official document, the community faq does not have this solution

I wont know if its really broken till 15 comes out, not really one to test the beta version of that.

arch1mede, 2021-09-22:
I found a much better solution where you don't have to make any changes to pikvm, look for the webtop docker

mdevaev, 2021-09-24:
I believe that we can hardly do anything here except for the advice to use a valid SSL certificate. You should try tyd without pekwm and address a question to the tyd developer if you find any other bugs: https://github.com/tsl0922/ttyd

mdevaev, 2021-09-24:
Please Let me know about this issue.

arch1mede, 2021-10-14:
Just FYI, this is also seen in Unraid as its also using ttyd 1.6.3. I was going to check proxmox but looks like they are using a different mechanism. This is not a PiKVM bug as this seems to only be seen in iOS15. Older versions do not exhibit the same behavior.

arch1mede, 2021-10-14:
Able to repo using OSX Safari. Using another browser works as expected.