pikvm/pikvm #433: Support for multiple ATX Modules

colindawson, 2021-08-29:
**Is your feature request related to a problem? Please describe.**
Currently the V3 implementation allows the ATX control to control a single ATX power supply.
When combining the PIKvm with a hardware KVM, it is not possible to power on and off all the individual machines.

**Describe the solution you'd like**
When using a KVM Like the Ezcoo there's an override that can be used to add input selection to the GPIO Menu, which works great.  This allows switching between 4 different machines (and other switches allow more) which radically enhances the PiKvm's abilities.

I'd love design a breakout board to allow selection of different ATX modules, so that rather than being limited to powering just one machine, each machine in attached to the KVM could be powered on and off via the ATX controls in the webpage.

The way that I would do this is to use the existing circuit to actually perform the ATX commands (this maintains backward compatibility, and keeps implementing a single ATX as easy as it is now)

I'd like to design a circuit board that allows selection of different channels for the ATX commands, so that **when I select an input for the Ezcoo KVM, the PiKvm switches to the correct atx module**.

I have not attempted to design the channel selection circuit yet, and have not designed a protocol for selecting the channels.  See the alternatives for the options that I've thought about so far.

**Describe alternatives you've considered**

Option 1.
Using two GPIO pins on the PI, to select the channel.   The two pins will be designated as "data" and "clock".
This will allow a serial byte array to be streamed out of the pi, allowing for as many channels to be created as desired.
In the case of a 4 port KVM, the GPIO pins will send a single bit representing the channel to activate.
0001, 0010, 0100, or 1000.  Flipping the clock pin from low to high, will set the current value on the data pin.

Pros

- this will allow a simple bit stream to select the active channel..
- can add as many channels as desired.  So not limited to 4 channels, could easily run 32 channels, which would mean a larger circuit to route the atx channel data.

Cons

- Timing has no feedback, so it's impossible to know if the sequence remains in sync.

Option 2
Could use multiple GPIO pins to represent a binary number.
The more PINS assigned the bigger the number.   1 pin would allow for 2 ATX's modules. 4 pins allowing 16 channels.

Pros

- this allows the selected channel to be a binary number.
- it's expandable, adding one more GPIO pin to the binary number doubles the amount of ATX Modules that can be supported.
- The selected channel will remain known, as the binary number will state with channel, this cannot get out of sync.

Cons

- Might end up using alot of precious GPIO Pins.

Option 3
Implement a serial port ATX channel selector

Pros

- unlimited number of channels possible.

Cons

- using a real serial port will add alot of complexity. 

Personally I like option 2 the best from these.  

**Additional context**
None at present, this is an idea and a work in progress

mdevaev, 2021-08-30:
Thank you for the ideas. We will definitely implement something for this.

gitneko, 2021-09-18:
I'd suggest to use I2C GPIO expander chips to implement this, so you don't use any GPIO pins of the Raspberry Pi itself (which could be used better). I2C is a common protocol - both supported hardware and software on the Raspberry Pi - and GPIO expanders are fairly easy to implement.

Ozzard, 2022-01-19:
I have a mild preference for a serial port interface here, selectable between 3-wire and USB, simply to increase portability of PiKVM.  My preferred Pi "GPIO" expander is a Pico, which I power from a USB port and can therefore use a minimal USB serial interface.  If I wanted to put it into a tidy box that exposed the Pi USB ports, such as a modified or extended v3 case, then I could instead use one of the Pi4's six(!) onboard serial ports, the Pico's built-in serial port, and make one configuration change in the Pico build script and one in the serial port assignment on the Pi, and everything would Just Work.

If you want to swap to a different host or GPIO board, a well-defined serial interface with a microcontroller such as an Arduino or Pico at the far end is simple to implement; and if you want to integrate this onto a "v4 HAT" then it's very difficult to argue with under $1 for a RP2040.  You don't even need flash with a RP2040 if you're using a Pi - just use two GPIOs from the Pi to implement the SWD "debugging" interface and load the "firm"ware straight into the microcontroller's RAM on startup.  ~25 GPIOs for a little over $1 if you don't mind a slightly uneven clock on the RP2040.

Ozzard, 2022-01-19:
(Happy to collaborate on a design for such a thing, by the way)

nCr78, 2022-03-16:
Hello! Is there any update on this or maybe an alternative?

I have 4 ATX that I need to control their power remotely through hardware. The alternative solution I'm thinking for now is with wireless relay switches (like Sonoff) that could be controlled through the raspberry wirelessly with a script (if there's an API for this), but, it's rather expensive if you want to do this safely plus it requires an installation before the wall plug to the PSU of the ATX.

A solution with a USB to Serial with a driver maybe? Even with limited capabilities e.g. only power on/off the ATX with a script that turns on/off power to raspberry's USB port.

johntron, 2022-05-13:
If you don't care about having Power / LED status or the ability to reset (as opposed to power off / on), it seems one could connect the power button leads to one machine, and the reset button leads to another. This way power button operates one machine's power button, and reset operates a second machine's power button (short-press?). 

Obviously this would only work for 2 machines, but I'm also wondering if changing the status indicator pins from input to output might be a simple software change and might help scale up to 4 machines (power button only with no statuses).

mdevaev, 2022-10-17:
No news yet, sorry.

colindawson, 2022-10-18:
I'm working on a design that will use three GPIO pins on the PI, to allow selection of ATX Modules on up to 8 machines.  I'm designing the idea for up to 8, but only plan on implementing 4 for myself.    My plan is to use a Raspberry Pico to select between 5 channels.  (hence 3 GPIO pins for selection)   Channels 1-4 will be directly selectable, channel 5 will be for an uplink, allowing a second Pico to be used for another 4 computers.  My plan is the current GPIO pins on the PI will still be used for control, just that it'll relay through the pico which will act as a switch.     Ideally I'll have the whole thing running off the 3.3v line on the PI, so won't need any external power or anything.   At least that's my plan, only thinking about it at the moment, haven't got to actually trying anything but think if should work.

mortenboldt, 2022-11-02:
This is something I've been hoping would appear for a long time. Let me know if there is anything I can do to help.

zappanaut, 2022-11-12:
Hi there, first time poster here :)
Maybe I missed it, but I would like to suggest using USB and a device that can talk to the i2c_tiny_usb device driver.
Originally this one: https://github.com/harbaum/I2C-Tiny-USB
But there is also a recent port to rp2040 (https://github.com/Nicolai-Electronics/rp2040-i2c-interface) or, if you are more into PICs, this one should work, too: https://fischl.de/i2c-mp-usb/ 
That should give a lot of opportunities by using i2c devices (gpio, temperature, ...) on the remote machine.
Regards!

tao-j, 2022-11-24:
Hi, I would advertise this: https://github.com/notro/pico-usb-io-board/pull/2. This is using a rpi pico.

Now it should work on any latest arch or fedora distribution.

Just configure the gpio in override.xml you can have any amount of connections as long as they can fit it into 20 or so pins.

zappanaut, 2022-11-25:
> Hi, I would advertise this: [notro/pico-usb-io-board#2](https://github.com/notro/pico-usb-io-board/pull/2). This is using a rpi pico.
> 
> Now it should work on any latest arch or fedora distribution.
> 
> Just configure the gpio in override.xml you can have any amount of connections as long as they can fit it into 20 or so pins.

I am testing this at the moment and had only some minor issues (https://github.com/notro/pico-usb-io-board/issues/3). My plan is to use one RP2040 board per server.

I am also using a KVM-switch, that is controlled via lirc and using a custom menu on the web frontend to select the server. Is there any way to keep the ATX controls, the gpio-boards and KVM switch state in sync?

mutsumi63, 2022-12-01:
https://github.com/akhlut/PiKVM_Parts/tree/master/USB

zappanaut, 2022-12-01:
For ones that are interested, I am working on a board using a RP2040-Zero with the aforementioned dln2 software.

[image]

mortenboldt, 2022-12-01:
> For ones that are interested, I am working on a board using a RP2040-Zero with the aforementioned dln2 software.
> 
> [image]

Any chance you'll be selling those, along with the software I take it you've written to control it?

zappanaut, 2022-12-01:
> Any chance you'll be selling those, along with the software I take it you've written to control it?

I will publish the KiCAD files and also a little "tutorial" after I tested the PCB.
The software simply consists of
- the (slightly patched) firmware from https://github.com/notro/pico-usb-io-board
- some udev scripts to automatically register the device and make it available to kvmd
- override.yaml for kvmd for all the assignments
- dln2 device drivers for the PiKVM distribution using dkms

I don't want to start a business with this and it should be pretty easy to solder. But if you don't feel confident and just need one or two populated boards I may send them to you at cost.
The PCBs are pretty inexpensive. I just ordered 5 for only 10 Euros (shipping included!).

One can control multiple boards using a custom menu in the Web-Interface. I did not have a look how to assign a board's GPIOs dynamically to the standard ATX menu (in conjunction with an external KVM-switch). So, at the moment only one board can be assigned to that menu exclusively.

META-DREAMER, 2022-12-09:
@zappanaut really looking to setup multiple ATX control with my pikvm setup, would love the KiCAD files and documentation on how you set it up once it’s ready!

ProjectInitiative, 2023-01-20:
@zappanaut also interested.

Rogue136198, 2023-01-24:
I just read through this thread since I am also looking for a way to support ATX control on multiple hosts. I had an idea of how to implement this really elegantly if a partnership was formed...

@level1wendell has his own KVM line. I'd love to get one to support him but he does not currently have one that fit my needs with them all being DP instead of HDMI... :(

Now here is the idea. Imagine if Wendell was willing to make a KVM that is designed for the PiKVM with HDMI, USB and the required hardware for ATX control switching. That would be amazing and I think it would be a fantastic partnership between an OS project and someone who is a major supporter of Open Source. I don't know if this is something that will ever happen but it's certainly something I would happily pay good money for.

level1wendell, 2023-01-24:
Dp to hdmi converters can work fine with the l1t kvm

https://mobile.twitter.com/tekwendell/status/1617968097809534977

virtual-machinist, 2023-01-24:
Will be doing a quad switch for the existing ATX boards using solid-state relays. The switch will be controlled using I2C and will require +5V power. The switch is designed to be very dumb, i.e. basically switches wires between a pair of 8P8C sockets. Currently waiting on a batch of relays for testing. If it works out, will post a link here.
BTW does anybody have a link to the ATX board gerbers that came with the PiKVM kickstarter?

colindawson, 2023-01-24:
I’m still thinking about doing this using a rpico.   Think of the tax control as a master and 4 tax connections as switched slaves, could even use a USB port on the pi to power and control, or use the GPIO on the pi to control, not sure if that yet.   The whole thing will be using components just like the v3 diy atx connections

virtual-machinist, 2023-01-26:
More ideas is always better. I personally don't like using Pico if it's not easily hand-solderable (think a custom design using a bare RP2040 chip) or ridiculously expensive (everything over $5 for a Pico is IMO a rip-off). An ideal design should use some stamp clone that can be directly soldered onto a board with isolated IOs + 8P8C socket.
Can't say my design is very cheap if using original SSRs from Mouser, but I've ordered them from AliExpress for a fraction of the price. They frequently turn out working perfectly well even if counterfeit.

zappanaut, 2023-01-29:
It took some time but i managed to set up a few repositories. You may want to use https://github.com/zappanaut/pikvm-usb-atx-ctrl as a starting point. The documentation is rather brief, so feel free to use the issue tracker there. ;)

Regarding the costs: Shipping included, I paid $13 for three RP2040-Zero boards on aliexpress and $11 for 5 PCBs from JLCPCB. If you are not very picky regarding the other components (especially the PhotoMOS relays) the BOM costs are rather low.

ProjectInitiative, 2023-02-02:
I built a slight alternative to @zappanaut's great project. https://github.com/ProjectInitiative/pikvm-atx for a single raspberry pi pico, multi-breakout board alternative. Admittedly I was not looking to solder any SMT components, so I just bought a dumb board and wrote some firmware for a pico, hope this is useful, and could potentially get added to the main project as an alternative! (still very much a WIP)

shiecldk, 2023-02-11:
> It took some time but i managed to set up a few repositories. You may want to use https://github.com/zappanaut/pikvm-usb-atx-ctrl as a starting point. The documentation is rather brief, so feel free to use the issue tracker there. ;)
> 
> Regarding the costs: Shipping included, I paid $13 for three RP2040-Zero boards on aliexpress and $11 for 5 bare PCBs from JLCPCB. If you are not very picky regarding the other components (especially the PhotoMOS relays) the BOM costs are rather low.

@zappanaut Can this project support 4 ports?

zappanaut, 2023-02-12:
@shiecldk I can't see why it should not work with 4 or even more. At the moment I am using only two, though.

shiecldk, 2023-02-12:
Thank you very much. @zappanaut 

I also created a ATX KVM repo based on  @zappanaut 's inspiration. This is done through additional GPIO only with limitation of nor using the HDD LEDs to support up to 4 ports. If you find out additional GPIO pins that can be used for PC2 - PC4, feel free to let me know. 
https://github.com/shiecldk/pikvm-gpio-atx-multiport

zappanaut, 2023-02-12:
@shiecldk Off the top of my head I can't help with the available IOs on the Pi, but there are various options to extend them seamlessly. At the moment I would opt for an RP2040 ;) It should be possible to control at least 4 PCs with just one RPi-Pico.

shiecldk, 2023-03-03:
@zappanaut Could I ask you how do you flash RP2040?

zappanaut, 2023-03-03:
@shiecldk I am using the picotool from here: https://github.com/raspberrypi/picotool
If you want to use dln2 firmware together with pikvm you also have to install the dln2-dkms device drivers on your rpi.

shiecldk, 2023-03-03:
Thanks. I found a way to flash the RP2040 on my Mac. Also, it looks like RP2040 is not hot-pluggable. I would need to re-flash the firmware if I accentanaccidentallytally unplug the RP2040 from Pik-KVM.

zappanaut, 2023-03-03:
I am a bit confused. Why should you have to re-flash the firmware? Using the `picotool load` command should write the firmware to the flash memory and not just the RAM. But I think this is a bit off topic, so if you want to discuss this further maybe you can open a ticket here: https://github.com/zappanaut/pikvm-usb-atx-ctrl

hiarcy, 2023-08-04:
@zappanaut I might have missed it so apologies to everyone if I'm repeating anything. Likewise, it can take a bit of trial and error in person for me to fully understand stuff like this; am I correct in understanding that you use a single board per client (i.e. PC with ATX pins), which then gets connected to a USB slot on the client itself or does it have to go straight to the HDMI switch? I've yet to deal with anything custom boards so I'll be looking into KiCAD and how to get my own boards next.

mutsumi63, 2024-04-01:
Maybe another option 
I disassembled the XH-HK4401 4-port KVM switch and found that the chip inside is CH444G. Upon checking the datasheet, I found that I can directly obtain the active signal by reading the voltage from the two pins in the CH444G.
This means that the active signal can be extracted from it.
This resolves the issue of sequence synchronization.

Tested the high state of the two pins, which are 3.3V and 3V respectively, with low state  close to 0V. The four active channels signal are 01, 11, 00 and 10.

The current issue is how to design an ATX module switcher based on this signal.

[image]

[image]
[image]

techlobo, 2024-08-12:
@mutsumi63 whilst that information is interesting, the issue with an approach based on it is that it only applies to the XH-HK4401 KVM switch (and requires the switch to be hacked).

Active channel can be determined via the PiKVM from this and (I believe) other switches e.g. Ezcoo.

techlobo, 2024-08-12:
Understood, and I wish you all success with your project!

I assume that you are only targetting the XH-HK4401

shiecldk, 2024-09-26:
@mutsumi63 @techlobo Here is the suggestion from AI (Perplexity.ai) for designing your ATX-multiport switcher. Hope it will help saving your development time.

https://www.perplexity.ai/search/read-this-thread-about-how-to-tZDZeVDfQy2uB7PBKWyhvA#2

To design a Multi-ATX Switcher that synchronizes with an HDMI switcher using the XH-HK4401 KVM switch, you will need to focus on several key components and steps. Below are detailed instructions, including the necessary hardware, components, and circuit design considerations.

## Overview of Design

The Multi-ATX Switcher will utilize the active channel signals from the CH444G chip in the XH-HK4401 to control multiple ATX power supplies based on the selected KVM channel. The design will include a microcontroller to read these signals and control relays or MOSFETs that manage power to the ATX units.

## Components Required

### 1. **Microcontroller**
   - **Raspberry Pi Pico**: A low-cost microcontroller with sufficient GPIO pins to read input signals and control outputs.
   - **Alternative Options**: Arduino Nano or ESP32 if you prefer different programming environments.

### 2. **Relay or MOSFETs**
   - **Relays**: Use 5V relays rated for the current required by your ATX units. Ensure they have a low activation voltage compatible with your microcontroller.
   - **MOSFETs**: If you prefer a solid-state solution, choose N-channel MOSFETs (e.g., IRF520) that can handle the load current of your ATX units.

### 3. **Power Supply**
   - A dedicated power supply for the microcontroller and control circuitry (5V).
   - Ensure that your ATX units have their own power supply that meets their specifications.

### 4. **Voltage Divider Resistors**
   - Use resistors to create a voltage divider if necessary to scale down voltage levels from the CH444G pins for safe reading by the microcontroller.

### 5. **Miscellaneous Components**
   - Diodes (e.g., 1N4001) for flyback protection when using relays.
   - Capacitors for decoupling and stabilization.
   - Connectors for inputs from the KVM switch and outputs to the ATX units.

## Circuit Design

### Signal Synchronization
1. **Reading Active Signals**:
   - Connect the two relevant pins from the CH444G chip to two GPIO pins on the microcontroller.
   - Implement a voltage divider if needed to ensure that voltage levels are within acceptable limits for the microcontroller.

2. **Control Logic**:
   - Write firmware for the microcontroller to read the GPIO pins at regular intervals or using interrupts.
   - Map the read values (binary states) to corresponding ATX outputs.

### Relay/MOSFET Control
1. **Relay Control Circuit**:
   - Connect each relay coil to a GPIO pin on the microcontroller.
   - Use diodes across relay coils for back EMF protection.
   
2. **MOSFET Control Circuit**:
   - Connect gate of each MOSFET to a GPIO pin on the microcontroller.
   - Ensure proper gate resistor values (typically 10kΩ) to pull down when not driven.

### Power Management
- Ensure that all components are rated for their respective voltages and currents.
- Use separate power supplies for logic (5V) and ATX units (typically 12V), ensuring common ground between them.

## PCB Design

1. **Schematic Creation**:
   - Use PCB design software like KiCAD or Eagle to create your schematic based on the above components and connections.

2. **PCB Layout**:
   - Place components logically, minimizing trace lengths between high-current paths and control signals.
   - Consider thermal management if using relays or MOSFETs that may generate heat during operation.

3. **Testing Points**:
   - Include test points in your design for easy debugging of signal inputs and outputs.

## Testing and Iteration

1. **Initial Testing**:
   - After assembling your PCB, perform initial tests with multimeters to check voltage levels at various points in your circuit.
  
2. **Functional Testing**:
   - Power up your circuit and test switching between KVM channels, verifying that corresponding ATX units turn on/off as expected.

3. **Adjustments**:
   - If any issues arise, revisit your schematic/layout, check connections, and modify firmware as necessary.

4. **Documentation**:
   - Document all aspects of your design process, including schematics, PCB layouts, firmware code, and testing results for future reference and community sharing.

By following these detailed steps, you can successfully produce a PCB and circuit for an ATX multiport switch that synchronizes with an HDMI switcher at a hardware level, eliminating the need for communication with PiKVM.

Citations:
[1] https://ppl-ai-file-upload.s3.amazonaws.com/web/direct-files/22904765/24277adb-4dd0-447d-a77f-e20b6e1824d4/XH-HK4401.pdf

shiecldk, 2024-10-10:
> Thank you very much. @zappanaut
> 
> I also created a ATX KVM repo based on @zappanaut 's inspiration. This is only done through additional GPIO pins with limitation of not using the HDD LEDs to support up to 4 PCs. If you find out additional GPIO pins that can be used for PC2 - PC4, feel free to let me know. https://github.com/shiecldk/pikvm-gpio-atx-multiport
> 
> You'll need to make a custom ATX switch based on the GPIO pins mapping and a HDMI USB 4-ports KVM switch.

Btw, I can confirm this is working on my hand-soldered PCB for 4PCs. If anyone is interesting in this project. We may work together on drawing this PCB and the BOM.

techlobo, 2024-10-11:
@shiecldk - glad you got it working.

However I am happy with my own setup ([ATX_Switch](https://github.com/techlobo/ATX_Switch/tree/main)) that supports all the current ATX functionality from the standard PiKVM UI. This approach should work with v2, v3 and v4+ PiKVM devices.

colindawson, 2024-10-11:
That looks excellent.    I might end up building one to put into my DIY PiKvm.    Beats the idea that I had that used a Pico.

shiecldk, 2024-10-16:
> @shiecldk - glad you got it working.
> 
> However I am happy with my own setup ([ATX_Switch](https://github.com/techlobo/ATX_Switch/tree/main)) that supports all the current ATX functionality from the standard PiKVM UI. This approach should work with v2, v3 and v4+ PiKVM devices.

Wonderful. Would this work on DIY PiKVM V2?

techlobo, 2024-10-16:
As v2 is DIY you should be able to configure it to work with either approach, provided it is based on Pi4.

My own DIY unit is effectively a v2+ (built on the v3 box image) as I wanted to have the PWM Fan, OLED and RTC functionality - however none of this is required for ATX Switch functionality.

I've never tried it with a zero2w, although _theoretically_ it should work provided you bring the relevant GPIO pins out to an RJ45 ATX_Switch compatible port (or adapt a cable to match the required pinout).

mdevaev, 2024-12-21:
After 3 years we have the official solution: [PiKVM Switch](https://docs.pikvm.org/switch/).

It is available in our [international store](https://shop.hipi.io/product/pikvm-switch-multiport-extender).
Canadian customers can place an order at [PiShop.ca](https://www.pishop.ca/product/pikvm-switch-multiport-extender/).

As usual, we welcome DIY, but this issue is closing because there is a solution :)

[image]

closed 2024-12-21 by mdevaev