pikvm/pikvm #436: pikvm password protect of access webinterface

Morta1985, 2021-09-01:
When i add this two lines in /etc/kvmd/nginx/nginx.conf

server{
..
auth_basic "Administrator Login";
auth_basic_user_file /etc/kvmd/nginx/.htpasswd;
}

The password prompt is coming but after that the normal login of pikvm doesn't work.
Why?

mdevaev, 2021-09-03:
Because pikvm has an alternative authorization method using http basic.

Why do you need this?

Morta1985, 2021-09-03:
because i have a reverse proxy which is directly in the internet

mdevaev, 2021-09-03:
forward pikvm backend without authorization, it has own

closed 2021-09-09 by mdevaev

mdevaev, 2022-01-04:
@hugomon noauth mode described in the faq

adamoutler, 2022-06-28:
Can you post a link to noauth mode? 
This is literally the only result on google.  
[image]

mdevaev, 2022-06-28:
> How do I completely disable authorization in PiKVM?

https://docs.pikvm.org/faq/#first-steps