pikvm/pikvm #448: Add http requests or mqtt support

connected201, 2021-09-07:
hi, i have loot of relay from sonoff or other vendors, and these relays can be started or stopped with simple http request or publish mqtt topic, with these relay i trigger to start or stop my PC
[image]
Sonoff SV

 it's possible to add option in ATX menu option to make http request to relay?

[example](https://github.com/xoseperez/espurna/wiki/RESTAPI)

mdevaev, 2021-09-20:
We need to write [GPIO](https://github.com/pikvm/pikvm/blob/master/pages/gpio.md) plugin for this. I don't have a device to make and test this.

connected201, 2021-09-22:
this rellay is triggered via MQTT protocol, no wires! this device have wifi module on board (esp8266)

philipforget, 2021-12-15:
This one would be really useful for me too, easy to tuck one of these away inside the case near the ATX connectors of the mobo vs running wires back to a central point. 

A board like this is [cheap 4 relay esp32 dev board](https://www.aliexpress.com/item/1005003161500835.html), or maybe even a cheap custom board with optoisolators and atx style connector passthrough could be designed for this purpose.

For the actual actuation, something like mqtt could work with a small broker container running on pikvm (or supply your own). Maybe leveraging something like https://esphome.io/ for building a small firmware and OTA updates etc. I'll investigate this a bit more in the new year too.

adamoutler, 2022-01-09:
Run Local Script tab would be helpful here.  Some predefined scripts could be activated, scheduled, and based on the output a command could be issued.   With this solution, #448, #460 ... too many to enumerate.  They could be addressed as well.

ndng28, 2022-01-16:
+1 for this feature request

mdevaev, 2022-11-10:
There is the CMD plugin to run scripts: https://docs.pikvm.org/gpio/#cmd

adamoutler, 2022-11-10:
Well, great!  Curl is an option for http requests.

Slarag, 2023-01-13:
CMD plugin and curl can be used to send out MQTT requests, but is there also an option to get feed back via MQTT? E.g. a relay is set via MQTT (either via a button in the PiKVM frontend or some other device/script interacting with MQTT) and this relay reports back its status via another MQTT topic. I would make sense, that the toggle button then reflects the value that was last reported back.

adamoutler, 2023-01-13:
Wrap a `netcat -l` in a `while` loop and parse the output?

Slarag, 2023-01-15:
What about a plugin for MQTT that wraps [asyncio-mqtt](https://pypi.org/project/asyncio-mqtt/)?

I have not really worked with async code yet, so please correct me if I'm getting something wrong about the plugin infrastrucure or other terms. Opinions/ideas/feedback on the idea of such a plugin?

Configuration could look like this:
```
kvmd:
    gpio:
        drivers:
            mqtt:
                type: mqttrelay
                broker: localhost
                port: 1883
                ...
        scheme:
                relay1:
                    driver: mqtt
                    mode: output # Or bidirectional or similar
                    switch: true
                    write_topic: "/relay1/set"
                    # Read topic is the topic where the relay publishes its status when it changes
                    read_topic: "/relay1/value"
                relay2:
                    driver: mqtt
                    mode: output # Or bidirectional or similar
                    switch: true
                    write_topic: "/relay2/set"
                    read_topic: "/relay2/value"
                ...
```

mdevaev, 2023-01-19:
The idea is ok, but we can't transfer custom options to the scheme right now. To do this, we need to make a lot of changes inside plugins, but I don't have time for that yet.