pikvm/pikvm #450: OTG Mouse not working on NVR/DVR CCTV etc...

antwal, 2021-09-08:
**Describe the bug**
I'm testing pi kvm to use it on various NVR / DVR; but the mouse doesn't seem to work here, it only allows left click in the center of the screen; there is no possibility to move the mouse on the screen.

**To Reproduce**
Currently I have only tested on Dahua and HikVision products

**Desktop (please complete the following information):**
 - OS: Apple OS X
 - Browser chrome, safari
 - Version 93.0.4577.63, 12.1.2

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform v2-hdmi
  - Video capture type CSI bridge
  - KVMD version: `kvmd 3.11-1
kvmd-fan 0.1-1
kvmd-oled 0.3-1
kvmd-platform-v2-hdmi-rpi4 3.11-1
kvmd-webterm 0.40-1`
  - uStreamer version: `ustreamer 4.4-1`
  - Linux kernel: `Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux`

NiKiZe, 2021-09-08:
A stab in the dark, have you tried relative mode for mouse?
https://github.com/pikvm/pikvm/blob/master/pages/mouse.md

antwal, 2021-09-08:
> A stab in the dark, have you tried relative mode for mouse?
> https://github.com/pikvm/pikvm/blob/master/pages/mouse.md

yes I have tried everything; if you need logs or anything else to understand the problem let me know.

NiKiZe, 2021-09-08:
This is most likely an issue with the Dahua/HikVision devices - can you get any logs from those for how the pikvm is detected? And does any part of keyboard interactions work?

mdevaev, 2021-09-09:
1. Try full system update, new kernel contains some HID fixes.
2. Try this: https://github.com/pikvm/pikvm/blob/master/pages/msd.md#disable-msd

antwal, 2021-09-09:
> This is most likely an issue with the Dahua/HikVision devices - can you get any logs from those for how the pikvm is detected? And does any part of keyboard interactions work?

access in ssh or telnet is very limited, I can send you very little information and I cannot tell you how pikvm is recognized, this is the information:

```
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  5 threads
  7 vcs
 10 misc
 13 input
 21 sg
 29 fb
 89 i2c
 90 mtd
128 ptm
136 pts
153 spi
180 usb
189 usb_device
204 ttyAMA
218 himedia
253 dh_poe
254 bsg

Block devices:
  1 ramdisk
259 blkext
  7 loop
  8 sd
 11 sr
 31 mtdblock
 65 sd
 66 sd
 67 sd
 68 sd
 69 sd
 70 sd
 71 sd
128 sd
129 sd
130 sd
131 sd
132 sd
133 sd
134 sd
135 sd

          CPU0       CPU1       
 33:      36495          0       GIC  33  tick0
 34:          0      35597       GIC  34  tick1
 37:          0          0       GIC  37  RTC Alarm
 38:          0          0       GIC  38  uart-pl011
 40:         16          0       GIC  40  uart-pl011
 46:          0          0       GIC  46  pl022
 48:      13317          0       GIC  48  100a0000.ethernet
 49:      10798          0       GIC  49  11010000.sata
 50:         24          0       GIC  50  ohci_hcd:usb2
 51:        198          0       GIC  51  ehci_hcd:usb1
 53:      17460          0       GIC  53  hi_vdec_vdh_irq
 55:      17460          0       GIC  55  hi_vdec_scd_irq
 62:       8671          0       GIC  62  VOU Interrupt
 63:          0          0       GIC  63  VEDU_0
 64:      11444          0       GIC  64  hi_tde_irq
 65:          0          0       GIC  65  IVE
 66:          0          0       GIC  66  hi_jpeg_irq
 67:          0          0       GIC  67  VGS0
 69:      16158          0       GIC  69  VPSS0
 70:          0          0       GIC  70  VOIE
 71:          0          0       GIC  71  JPEGU_0
 72:      10011          0       GIC  72  AIO Interrupt
 75:          0          0       GIC  75  VDA
 93:        681          0       GIC  93  100a0000.ethernet
 94:      14964          0       GIC  94  100a0000.ethernet
 95:       8991          0       GIC  95  100a0000.ethernet
IPI0:          0          1  CPU wakeup interrupts
IPI1:          0          0  Timer broadcast interrupts
IPI2:      54374      98350  Rescheduling interrupts
IPI3:          0          0  Function call interrupts
IPI4:          0       7589  Single function call interrupts
IPI5:          0          0  CPU stop interrupts
IPI6:          0          0  IRQ work interrupts
IPI7:          0          0  completion interrupts
IPI8:          0          0  CPU backtrace
```

the mouse that comes with both is model is RTM 019; I have KVMs from Aten and these work great.

antwal, 2021-09-09:
> 1. Try full system update, new kernel contains some HID fixes.
> 2. Try this: https://github.com/pikvm/pikvm/blob/master/pages/msd.md#disable-msd

I try to carry out this test and update you about it

antwal, 2021-09-09:
1. I have updated:

**Pi-KVM info:**
  - KVMD version: `kvmd 3.23-1
kvmd-fan 0.1-1
kvmd-oled 0.3-1
kvmd-platform-v2-hdmi-rpi4 3.23-1
kvmd-webterm 0.40-1`
  - uStreamer version: `ustreamer 4.5-1`
  - Linux kernel: `Linux pikvm 5.10.52-12-ARCH #1 SMP Wed Sep 1 05:47:56 MSK 2021 armv7l GNU/Linux`

2. i disabled mass storage

the problem is not solved.

mdevaev, 2021-09-09:
Try relative mouse now after the update.

antwal, 2021-09-09:
> Try relative mouse now after the update.

yes now it works, but the mouse has a lot of lag, both with msd enabled and not

mdevaev, 2021-09-09:
How many lags you see?

antwal, 2021-09-10:
> How many lags you see?

if I can I try to make a video to evaluate the mouse lag; but to give an idea let's say that it seems to value on a remote desktop where the connection has 400/500 ms of delay

mdevaev, 2021-09-12:
Try to enable `Squash mouse moves` in web ui menu `System`

antwal, 2021-09-15:
> Try to enable `Squash mouse moves` in web ui menu `System`

Hello,

sorry for the delay but it was no longer possible for me to rehearse; I hope to succeed these days. I have tried various combinations on these settings but the problem persists

[image]

Hello,

sorry for the delay but it was no longer possible for me to rehearse; I hope to succeed these days. I have tried various combinations on these settings but the problem persists

Otherwise everything seems to work, even the keyboard. The only problem would be this of the mouse and to make the drive work which for the latter thing is better for me in case to open another issue

Thanks.

mdevaev, 2021-09-17:
Does a hardware mouse work normally?

antwal, 2021-09-17:
> Does a hardware mouse work normally?

yes without any problem, it is very fluid. 
I have also tried with some mouse that have the wireless usb key and even those have no problems.

[mouse test video on dahua nvr](https://youtu.be/Tv8-P8ZpiaQ)

if you move slowly, with precision waiting for the mouse to update, you can use it without problems.
if i can find some time, i also create the video on a hikvision nvr, but the problem is the same

mdevaev, 2021-09-18:
Okay, I know what this might be related to, but I'm not sure how I can solve it, because I don't have my own text environment. Most likely, this is due to the mouse update rate being too frequent. The USB mouse protocol is actually polling - that is, the host asks the mouse about what is happening to it, and the mouse responds with new coordinates or does not respond anything. Hardware mouse drivers usually have a rate of 8 ms, while in the kernel this value is 4. It is possible that the host driver does not handle this correctly. If you wait a couple of days, I will find time to do a test build of the kernel with a different frequency and we will see if this solves the problem.

antwal, 2021-09-18:
No problems, I am doing other tests also on raspberry pi zero

before using pikvm i used an arduino hid mouse based on the MouseTo project and with that it worked fine without delay if that can help

johnheusinger, 2021-11-16:
Hello - Also facing a similar issue here.

On an Amcrest NVR, mouse clicks are registering but the cursor won't move. In my case a hardware mouse also works fine.

I'd be interested to try adjusting the polling frequency if you think that'll help. Cheers

dustymabe, 2022-02-16:
Seems like issues with NVR systems isn't uncommon: https://www.reddit.com/r/pikvm/comments/oc8x57/comment/ho0vinj/?utm_source=share&utm_medium=web2x&context=3

There is an offset with my mouse so bad it's not really workable.

kevinmarchant, 2022-07-26:
> Okay, I know what this might be related to, but I'm not sure how I can solve it, because I don't have my own text environment. Most likely, this is due to the mouse update rate being too frequent. [...]

@mdevaev - Did the test build to hopefully help diagnose / support Reolink (and similar) mouse movements ever happen?  I’ve just recently got a PiKVM Hat 3 and as observed by others the mouse support still doesn’t work unfortunately.

mdevaev, 2022-07-29:
I made a test kernel, but to be honest, I don't think it will help. As with other similar problems, I need to look at the protocol dump in the cable to understand what's the matter. Although let's try it anyway.

https://files.pikvm.org/tmp/
Before installing this build, you need to completely update the OS. Then download it on PiKVM and install as follows:
```
pacman -U linux-*.pkg.tar.xz
reboot
```

kevinmarchant, 2022-07-29:
@mdevaev 

Tested and as you say, no difference.  

For a while after doing the preparation (pacman -Syu / reboot) and before applying the kernel updates, I had thought I saw significant improvement just from the recent kvmd / kvmd-platform* updates (I had only updated my system a few days ago so surprised to see any updates at all).  It was the smoothest mouse movements I had ever seen on my NVR - as good as when testing against a PC.  I took a backup at this point before updating the kernel patches.

The kernel patches did not help so I reverted back to the backup - unfortunately the poor performance returned as well and I have not been able to recreate the situation where it was working well earlier this afternoon.

For info, the mouse will only move if in relative mouse mode, which also means it won't move in VNC.  I've tried explicitly setting the mouse OTG to the details of the wired optical mouse supplied with the device - in my testing against a Windows machine I noted that even though set just as a mouse this did NOT disable the keyboard.  The NVR does not support a connected keyboard, just a mouse and I wondered if the presence of a keyboard might be causing confusion.

If there is any way I can gather the relevant logs then I'm happy to help.

```
#    # See https://github.com/pikvm/pikvm/blob/master/pages/vnc.md
#    keymap: /usr/share/kvmd/keymaps/ru # Set russian keymap
#    auth:
#        vncauth:
#            enabled: true # Enable auth via /etc/kvmd/vncpasswd

kvmd:
    hid:
        mouse:
            absolute: false
    msd:
        type: disabled
    atx:
        type: disabled
otg:
    manufacturer: Silicon Labs
    product: USB Input Device
    vendor_id: 0x01c4
    product_id: 0x8105
    serial:

```

mdevaev, 2022-07-29:
This is too low-level a problem, we have some logs, but it will be very difficult to collect them and I'm not sure what is useful. KVMD does not affect the mouse in any way in our case, it's all about the kernel and the low-level USB things.

mdevaev, 2022-10-07:
If someone could donate (or for a small money) me a similar device (old, almost broken, etc., it doesn't matter, as long as I can reproduce the problem with the mouse) and send it to me by post to Cyprus, I will look at it through the USB protocol analyzer. I can't guarantee I'll fix the problem, but at least I'll try.

mdevaev, 2022-10-08:
A similar issue with absolute mouse: #829 829

kevinmarchant, 2022-10-08:
Haven’t got a dead NVR but would be willing to crowd fund €10 towards buying you a new one.  How many of us are there and how much does the cheapest Reolink NVR cost n Cyprus ?  Can you buy on from Amazon and return it within the returns window at no charge to anyone ?

mdevaev, 2022-10-08:
I would prefer an old device from ebay :)

hj0x3, 2023-03-22:
Seems I was able to get this work now with a reolink NVR by enabling relative mouse and disabling horizontal scrolling.

https://docs.pikvm.org/mouse/#dual-mode

dustymabe, 2023-03-28:
I'm able to confirm that enabling relative mode and disabling horizontal scrolling works for me too:

```
kvmd:                       
    hid:                     
        # https://github.com/pikvm/pikvm/issues/450#issuecomment-1480385542
        mouse:            
            absolute: false
            horizontal_wheel: false
        mouse_alt:           
            device: /dev/kvmd-hid-mouse-alt
```

mdevaev, 2023-03-29:
@antwal could you test this solution?
```yaml
kvmd:
    hid:
        mouse:
            absolute: false
            horizontal_wheel: false
```

kevinmarchant, 2023-03-29:
@mdevaev It does seem to be working much better with the "horizontal_wheel: false" configuration.  Since last I contributed, I've fully updated the PiKVM software (pacman -Syu etc etc) and am using this override.yaml file which specifies the vendor_id / product_id of the mouse that came with the NVR.

I'd say the user experience is "good enough" - it's still slightly sluggish in my environment (my Pi is connected using wifi) but I've never had it working as well as this.  It won't replace the NVR web interface but does allow you to remotely reboot and configure some settings that can only be done at the console,

Whether the improvement is solely down to the "horizontal_wheel" or in conjunction with other software updates I applied is unclear, but thanks - I think I've got something I can use now.

[override.yaml.txt](https://github.com/pikvm/pikvm/files/11104065/override.yaml.txt)

mdevaev, 2023-04-23:
Well, apparently this is quite workable, so I put it in the FAQ. I suppose the issue can be closed. Let me know if anyone think something can be improved here.

closed 2023-04-23 by mdevaev

mdevaev, 2023-11-27:
@cyberopal did you try to disable mass storage?

cyberopal, 2023-11-27:
please advice where and how

mdevaev, 2023-11-27:
https://docs.pikvm.org/msd/#disabling-mass-storage

cyberopal, 2023-11-27:
@mdevaev sorry no change.... i disabeld the mass storage but still nothing