pikvm/pikvm #451: Discord invitation doesn't seem to work

e-minguez, 2021-09-08:
**Describe the bug**
Discord invitation doesn't seem to work

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to the README
2. Click on the discord invitation

**Expected behavior**
It doesn't work

**Screenshots**
[image]

**Desktop (please complete the following information):**
Irrelevant I guess

**Pi-KVM info:**
N/A

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-09-09:
Just checked, everything works. Probably some kind of temporary glitch. Try again.

closed 2021-09-09 by mdevaev

e-minguez, 2021-09-09:
You are right, it works now, thanks!

mdevaev, 2021-09-09:
Ur welcome