pikvm/pikvm #452: Reboot after the first boot

e-minguez, 2021-09-09:
Currently the preparation script doesn't reboot the Pi but instead just leave the OS read only. For remote sites it would be nice if it reboots so you can basically zero touch provision it.

Probably some setting such as `REBOOT=true` in the pikvm.txt file or something similar, then check for that setting in the `_pikvm-bootconfig.sh` script in order to reboot it or not.

e-minguez, 2021-09-09:
Reading this https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md#the-final-steps

It says:

```
After power-up, Pi-KVM OS generates unique SSH keys and certificates. Do not turn off the Raspberry Pi until it's fully booted. If you set up Wi-Fi in step 1, it won't be able to connect to the network on the first boot. You need to wait 10 minutes for all the OS preparations to complete, then just restart the device. On the second boot, the Raspberry will connect to the network without any problems.
```

So instead of waiting 10 minutes and then restart the device manually by the user, just let the preparation script to reboot the system.

In my case I'm trying to set this with a pi4 with no keyboard/screen attached, so there is no way to check when the 1st preparation step ends.

mdevaev, 2021-09-09:
I remembered why I didn't do this - the problem is that systemd does not survive the reboot call well at the boot stage here. But I don't remember the reason.

mdevaev, 2021-10-16:
Done. Will be available in the next image release.

closed 2021-10-16 by mdevaev