pikvm/pikvm #459: Insignia VGA to HDMI Adaptor Displays No Signal

ossiemoore, 2021-09-09:
**Describe the bug**
When connecting a PiKVM to the HDMI output of the Insignia VGA to HDMI Adaptor, "No Signal" is displayed. (see attached screenshot)

Device:  Inignia - VGA to HDMI Adapter
Model: NS-PCAVH
Linke to Item on BestBuy: https://www.bestbuy.com/site/insignia-vga-to-hdmi-adapter-white/6418344.p?skuId=6418344

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect VGA to HDMI Adaptor to HDMI port of computer/laptop and Pi running PiKVM
2. Login to pikvm.
3. Observe No Display  video. (example screen shot attached)
4. SSH into pikvm and run `journalctl -u kvmd | tail -n 50` and observe output similar to the following:
```
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.584    stream] -- Device fd=15 opened
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.584    stream] -- Using input channel: 0
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.590    stream] -- Got new DV timings: resolution=1920x1080, pixclk=148500000
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.610    stream] -- Using resolution: 1920x1080
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.610    stream] -- Using pixelformat: UYVY
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.610    stream] -- Querying HW FPS changing is not supported
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.610    stream] -- Using IO method: MMAP
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.623    stream] -- Requested 5 device buffers, got 5
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [155188.624    stream] -- Unable to start capturing: Invalid argument
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.624    stream] -- Device fd=15 closed
Sep 09 13:42:27 pikvm kvmd[421]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [155188.624    stream] -- Sleeping 1 seconds before new stream init ...
```

**Expected behavior**
Display to show normally.

**Screenshots**

Screenshot from HP ZBook:

[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser: Firefox , Safari
 - Version: 91.0.2, 14.1.2
 - VNC client (if used): None

**Pi-KVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - Pi-KVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Output from `journalctl -u kvmd` command appears above.

arch1mede, 2021-09-11:
Please join the discord for troubleshooting this. Thanks.

mdevaev, 2021-09-12:
Some VGA converters do not respect the EDID that the video capture device provides. I recommend using a different converter.

mdevaev, 2021-09-12:
Also you can decrease resolution: https://github.com/pikvm/pikvm/blob/master/pages/edid.md

closed 2021-09-12 by mdevaev

wcck, 2021-12-16:
Hi @mdevaev 

Excuse me, Do you recommend VGA converters except https://www.aliexpress.com/item/4000553298530.html ?
Thank you.

arch1mede, 2021-12-16:
> Hi @mdevaev
> 
> Excuse me, Do you recommend VGA converters except https://www.aliexpress.com/item/4000553298530.html ? Or others solution to solve this problem? Thank you.

Please join Discord and review the Discord pins, there are some vga adapters that have been vetted.

wcck, 2021-12-17:
Hi @arch1mede 

Thank you for your response.
I had joined discord and search this solve I will try that.
Thank you.

wnark, 2022-08-29:
[image]
When your resolution is constantly changing, you can use this device, the Chinese name is "HDMI锁屏宝"：hdmi lock screen treasure
it can lock the resolution at a certain value, such as 1080P, so that VGA TO HDMI will not be unusable due to constant switching of resolutions.
@mdevaev

mdevaev, 2025-09-04:
I'm not sure for 100%, but this one should work: https://www.startech.com/en-us/audio-video-products/vga2hdu