pikvm/pikvm #460: Support for power on/off via smart power strip

dustymabe, 2021-09-09:
**Is your feature request related to a problem? Please describe.**

This may already exist and I just don't know about it. It would be nice if, as an alternative to ATX power control, we can have the pikvm control a smart plug/power strip. For example, the Digital Loggers [Web Power Switch](https://dlidirect.com/products/new-pro-switch). Opening up some devices to get access to the motherboard is a pain and controlling power at the source might be a nice option. 

**Describe the solution you'd like**

It would be nice if control of a remote power strip was possible and also integrated into the UI like is done for ATX power control. The ConsolePi has [support for power control](https://github.com/Pack3tL0ss/ConsolePi/blob/master/readme_content/power.md) like this and maybe we can take  some inspiration from there.

RyanWor, 2021-12-11:
Agree this would be a great idea. I use similar plugins for my 3D Printers that work with the cheap TPLink/Kasa smart plugs. I currently use PiKVM for remote control of a laptop that requires Bitlocker PIN at boot. Sometimes the device auto-reboots and then hits the timeout at PIN screen and goes to full off. I found a way in BIOS to enable a setting that powers on the laptop at AC connect so I am using a smartplug to interrupt the charger to force a power on from full off remotely. I would think it would be pretty simple to add a button in the menus of the web GUI that fires of a shell script to something like a TPLink plug on static DHCP or even to something like HomeAssistant.

adamoutler, 2022-01-09:
Seems like it would be good addition to have a KVM Script tab to allow curl, flip GPIOs,  and schedule pings to servers at a regular interval to detect network problems which could be corrected by a curl command on failure.

rhashcode, 2023-10-18:
any update on this ? https://python-kasa.readthedocs.io/ is great and I've been trying to integrate it

mdevaev, 2023-10-20:
No, I'm busy with other tasks

rinseaid, 2023-11-02:
I have a very ugly implementation to allow me to control my Kasa Smart Strip through PiKVM, using a crude bash script and messing around with GPIO pins. Essentially, I am telling PiKVM to use a specific GPIO pin as an input, and then changing them to outputs and writing desired values to them. Nothing is nor should be attached to those physical pins. This is a crude method of storing the real Kasa outlet state and displaying it in the PiKVM Web UI. I don't use ATX control so not sure if it breaks it, if you are then you need to ensure the GPIO pins don't conflict. 

Again, this is ugly, crude, and probably a bad idea. You can remove all the GPIO stuff if you don't care about seeing the outlet states within the web UI.

The next bad idea is you need to override pip to allow installation of the kasa module:
`pip --break-system-packages install python-kasa`

Reminder: this is a bad idea. Don't do it. If you do, understand that you're messing around with the expected environment of PiKVM and you should neither request nor expect any support after doing this.

Next, configure kvmd with commands for each of the outlets you want to control. My outlet strip has 6 outlets, but I'm using it with an ezcoo 4-port KVM so I'm only configuring 4. I don't believe this can be parametrized any further as per https://github.com/pikvm/pikvm/issues/983.

The pin numbers for the outlet[x]_on/off under the 'scheme' section are arbitrary. No need to change those. The pin numbers for outlet[x]_status do need to match a real GPIO pins (BCM numbering). Follow instructions [here](https://docs.pikvm.org/gpio/#raspberrys-gpio) to pick unused pins.

```
# cat /etc/kvmd/override.d/kasa.yaml
kvmd:
    gpio:
        drivers:
            outlet1_on:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 1, on]
            outlet2_on:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 2, on]
            outlet3_on:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 3, on]
            outlet4_on:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 4, on]
            outlet1_off:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 1, off]
            outlet2_off:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 2, off]
            outlet3_off:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 3, off]
            outlet4_off:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 4, off]
        scheme:
            outlet1_on:
                driver: outlet1_on
                pin: 0
                mode: output
                switch: false
            outlet2_on:
                driver: outlet2_on
                pin: 1
                mode: output
                switch: false
            outlet3_on:
                driver: outlet3_on
                pin: 2
                mode: output
                switch: false
            outlet4_on:
                driver: outlet4_on
                pin: 3
                mode: output
                switch: false
            outlet1_off:
                driver: outlet1_off
                pin: 6
                mode: output
                switch: false
            outlet2_off:
                driver: outlet2_off
[... 73 lines cut ...]
    else
      gpio -g write $CUR_GPIO_OUTLET_PIN 1
    fi
  fi
done
```

Configure views to utilize the new buttons and inputs, here's my example merged with my ezcoo config:
```
# cat /etc/kvmd/override.d/view.yaml
kvmd:
    gpio:
        view:
            header:
                title: Device Control
            table:
                - ["#Port 1", ch0_led, "ch0_button|KVM", outlet1_status, "outlet1_off|confirm|Off", "outlet1_on|confirm|On"]
                - ["#Port 2", ch1_led, "ch1_button|KVM", outlet2_status, "outlet2_off|confirm|Off", "outlet2_on|confirm|On"]
                - ["#Port 3", ch2_led, "ch2_button|KVM", outlet3_status, "outlet3_off|confirm|Off", "outlet3_on|confirm|On"]
                - ["#Port 4", ch3_led, "ch3_button|KVM", outlet4_status, "outlet4_off|confirm|Off", "outlet4_on|confirm|On"]
```

After a kvmd restart, all the GPIO states are reset and will display as off in the PiKVM web UI. The next time you turn on or off an outlet using the buttons in the web UI, each of the outlet states is updated. I usually just click one of the 'On' buttons and it updates within a few seconds. There are other ways around this such as a systemd unit that runs after kvmd, or perhaps using a timer, but I personally didn't need it so didn't bother.

USE AT OWN RISK. Should you make the unwise decision to utilize the above, test it thoroughly with your setup before plugging anything important into the Kasa outlets. There is no support or warranty expressed or implied. If it turns off your devices unexpectedly, drives your electricity bill through the roof, or sets your house on fire, you're SOL.

kaystrobach, 2024-04-25:
thought about having webhooks, which can be connected to homeassistant etc.

slendermon, 2024-10-05:
I made a script to the kasa outlet as well, similar to [rinseaid](https://github.com/rinseaid) and [kaystrobach](https://github.com/kaystrobach), but I want to make it clear in case somebody in the future needs help with their Kasa power strip

In Home Assistant Settings go to automations and scenes:
[image]

Now click "create automation", then "create new automation", then when adding triggers, scroll down, click "other triggers", scroll down, click "webhook", and change webhook id to whatever (it is like a password).
Then Do "Device",  Choose "tp link power strip (not the outlets)", pick "toggle" option for outlet name. 
[image]

##Try doing script first before reloading the override.yaml or it won't work
/etc/kvmd/kasa.sh
```
#!/bin/sh

weburl="https://homeassistant.example.com/api/webhook/"

hook1=[WEBHOOK-ID]
hook2=[WEBHOOK-ID]
hook3=[WEBHOOK-ID]
hook4=[WEBHOOK-ID]
hook5=[WEBHOOK-ID]
hook6=[WEBHOOK-ID]

repress() { # This function is to basically turn On and Off
    press="curl -X POST "$weburl$hook"" # Send a POST with curl to home assistant
    $($press) # Toggle Once
    $($press) # Toggle Again
}

if [ ! "$1" ]; then # IF the first argument is missing...
    echo "Kasa Script"
    echo "./kasa.sh [outlet number]"
elif [ "$1" == 1 -o "$1" == 2 -o "$1" == 3 -o "$1" == 4 -o "$1" == 5 -o "$1" == 6 ]; then # ELSE IF the argument equals 1,2,3,4,5,6
    echo "Running Hook $1"
    hook_var="hook$1" # Make a new variable name with $1 argument
    hook=${!hook_var} # Interpret the variable's value into a new variable called hook
    repress # Call the "repress()" function
fi
```

##I did not know I didn't need "pins" for the "scheme" yaml, surprised but I removed them and they work fine
/etc/kvmd/override.yaml
```
kvmd:
    gpio:
        drivers:
            outlet1:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 1]
            outlet2:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 2]
            outlet3:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 3]
            outlet4:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 4]
            outlet5:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 5]
            outlet6:
                type: cmd
                cmd: [/etc/kvmd/kasa.sh, 6]
        scheme:
            outlet1:
                driver: outlet1
                mode: output
                switch: false
            outlet2:
                driver: outlet2
                mode: output
                switch: false
            outlet3:
                driver: outlet3
                mode: output
                switch: false
            outlet4:
                driver: outlet4
                mode: output
                switch: false
            outlet5:
                driver: outlet5
                mode: output
                switch: false
            outlet6:
                driver: outlet6
                mode: output
                switch: false
        view:
            header:
                title: Switches
            table:
                - ["#Kasa Smart Power Strip"]
                - []
                - ["#Outlet1 #1:", outlet1]
                - ["#Outlet2 #2:", outlet2]
                - ["#Outlet3 #3:", outlet3]
                - ["#Outlet4 #4:", outlet4]
                - ["#Outlet5 #5:", outlet5]
                - ["#Outlet6 #6:", outlet6]
```

##It should look like this, and work after a `systemctl restart kvmd`
[image]

##Btw `journalctl -e` helped a lot with debugging when something didn't work while i was figuring out override.yaml

Ryan5tewart, 2025-05-07:
Wanted to add my two cents as well. There is a much much safer and easier standalone implementation of this. (works locally, doesnt require home assistant or webhooks or other integration) Building off of what those above me have already shared, you can also just use Node.JS and the tplink api to call them locally:

 **Install Node.js and npm:**
`sudo pacman -S nodejs npm --noconfirm`

**Install Kasa API Library - This lets scripts control the plug via Node.js.**
`sudo npm install -g tplink-smarthome-api`

**Confirm it's installed:**
`npm list -g tplink-smarthome-api`

**After installing the library, optionally use the built-in discovery tool to find each device's IP address**
`npx tplink-smarthome-api discover`

**You can use the following .sh scripts anywhere and create custom GPIO menus in the same method as above: (e.g., /etc/kvmd/):**
**Just add your IP address for each outlet you need,. (eg.  ` client.getDevice({ host: '192.168.0.0' })`)**, and run the script. no other integrations required, it connects to the device over LAN

I have only tested with a single Kasa smart-outlet, not a power-strip. Though i suspect it would be similar to configure. (Make sure that you configure the device to allow third-party support in the KASA app)

### 🔹 plug-on.sh
```
#!/bin/bash
set -e

export NODE_PATH=$(npm root -g)

node -e "
  const { Client } = require('tplink-smarthome-api');
  const client = new Client();
  client.getDevice({ host: '<OUTLET-IP>' })
    .then(device => device.setPowerState(true))
    .then(() => console.log('Plug turned ON'))
    .catch(err => {
      console.error('Failed to turn on plug:', err.message);
      process.exit(1);
    });
"
```

### 🔹 plug-off.sh
```
#!/bin/bash
set -e

export NODE_PATH=$(npm root -g)

node -e "
  const { Client } = require('tplink-smarthome-api');
  const client = new Client();
  client.getDevice({ host: '<OUTLET-IP>' })
    .then(device => device.setPowerState(false))
    .then(() => console.log('Plug turned OFF'))
    .catch(err => {
      console.error('Failed to turn off plug:', err.message);
      process.exit(1);
    });
"
```

### 🔹 plug-restart.sh
```
#!/bin/bash
set -e

export NODE_PATH=$(npm root -g)

node -e "
  const { Client } = require('tplink-smarthome-api');
  const client = new Client();
  (async () => {
    try {
      const device = await client.getDevice({ host: '<OUTLET-IP>' });
      console.log('Turning OFF plug...');
      await device.setPowerState(false);
      await new Promise(resolve => setTimeout(resolve, 2000));
      console.log('Turning ON plug...');
      await device.setPowerState(true);
      console.log('Plug has been restarted.');
    } catch (err) {
      console.error('Failed to restart plug:', err.message);
      process.exit(1);
    }
  })();
"
```

adamoutler, 2025-05-07:
You might try reducing that. People generally don't need all that for tplink plugs

`tplink-smarthome-api send --host <plug_ip> '{"system":{"set_relay_state":{"state":1}}}'`
Or even simpler 
`echo -n '{"system":{"set_relay_state":{"state":1}}}' | nc <plug_ip> 9999`

Ryan5tewart, 2025-05-07:
> You might try reducing that. People generally don't need all that for tplink plugs
> 
> `tplink-smarthome-api send --host <plug_ip> '{"system":{"set_relay_state":{"state":1}}}'` Or even simpler `echo -n '{"system":{"set_relay_state":{"state":1}}}' | nc <plug_ip> 9999`

That used to work on older firmware, but most modern Kasa plugs need encrypted commands. The Node API handles all that

Ryan5tewart, 2025-06-17:
> Anyway to control any smart plug via bluetooth? Because i want to power cycle a router, but that router is the WIFI access point, so if it's freezed or crashed, no WIFI of course.

How do you intend to maintain a connection with PiKVM, assuming that it's connected through the same router?   I cannot think of any smart plug that supports bluetooth, but this would be far more difficult to implement that youre better off just bridging two routers for redundancy.