pikvm/pikvm #468: Adding support for kvmd to have hdmi capture card support for web interface video

shubhamrait, 2021-09-14:
**Is your feature request related to a problem? Please describe.**
When using pikvm the biggest problem I have noticed when using it as a crash cart is lack of direct hdmi capture card support. What i mean by this is that the kvmd web interface on pikvm can have support to have the video from the other machine directly displayed in the video tag directly from the hdmi capture card itself rather than being streamed from raspberry pi. That would make things better in terms of better battery life for raspberry pi, ability to use raspberry pi zero w or such lower spec raspberry pi and still get better performance, highest FPS and also having off the shelf cases as well. Thus I am recommeding for pikvm to also have a crash card mode.

**Describe the solution you'd like**
The solution is relatively simple. In the kvmd web interface all that  needs to be done is make a 3rd option available as "crash cart/direct hdmi capture" along with "MJPEG" and "WEBRTC" option or make a seperate branch with only hdmi capture card. In the master branch of kvmd you would have to add additional radio-box option in kvmd/web/kvm/navbar-system.pug  and thus in kvmd/web/kvm/index.html . Then after you will have to add some additional code in stream.js file so that the user has a choice to select hdmi card as input. It will need the code to access webcam as hdmi capture card appears as a webcam to a laptop.

**Describe alternatives you've considered**
There are no alternatives possible at this time

**Additional context**
I am working on it. My javascript is not great and I am trying.

shubhamrait, 2021-09-14:
The function for hdmi capture card streamer would be based upon _JanusStreamer() function in stream.js .

mdevaev, 2021-09-16:
I don't quite understand how this should work and how you expect to transfer video from another host. What is a crash card?

shubhamrait, 2021-09-17:
> 
> 
> I don't quite understand how this should work and how you expect to transfer video from another host. What is a crash card?

Crash cart is a KVM designed to operate locally. https://www.amazon.com/IOGEAR-Portable-Console-DisplayPort-Adapters/dp/B073PFLFGR shows an example. What my idea was that we can hve pikvm's web interface to include a mode that allows direct video display of the machine connected using hdmi acpture card.

arch1mede, 2021-09-26:
Community member made a crash cart, you can read about it [here](https://docs.google.com/document/d/1M9xUgNE_-P8GydKr_3qIgXUR9YzqApHNPhetRv3pcsE/edit?usp=sharing) and [here](https://docs.google.com/document/d/1wgBZHxwpbJWkJBD3I8ZkZxSDxt0DdNDDYRNtVoL_vK4/edit?usp=sharing).

mdevaev, 2021-09-28:
Thanks for the explanations.

It is impossible to display an image in the browser from a local computer without the participation of streaming code. Even if it is a local video capture device, you will have to run a streamer. The browser has an API to demonstrate a local camera, but it's not a fact that this will work with a video capture device, and it requires special settings - for example, HTTPS is mandatory.

As far as I understand, you want to reduce latency, but this will not work. This would be possible if you had a separate local interface (not the web) that uses KVM and a local video capture device.

This is possible, but goes far beyond the scope of PiKVM. I advise you to use it as it is, because you can already use PiKVM as crash cars, even if connected via a short network cable to a laptop. If someone makes a patch for this and this patch does not change the logic of the web interface too much, I will *probably* accept it. But I will not implement this myself, since I have more priority tasks. Sorry.

closed 2021-09-28 by mdevaev