pikvm/pikvm #470: Ask for supporting VIDEO OUTPUT from maybe HDMI output on Pi

Fisherworks, 2021-09-15:
**Is your feature request related to a problem? Please describe.**
If a local user can still view (or even control) the machine (display) that connected on the Pi-KVM which can simultaneously controlled by remote admin, this will be super helpful for our field engineers who can connect the KVM on customer sites and enable connectivity for remote diagnostic experts while the local customer/engineer can still view what's going on. 

**Describe the solution you'd like**
Use (one of) the HDMI outputs of Pi as video output as mirror of the video input.

**Describe alternatives you've considered**

- Design a specific video (HDMI/VGA/DP) output on the HAT.
- Or still use the Pi HDMI output, but the video output maybe wrapped by an Linux application window, is still much better than no output. 

**Additional context**
Or maybe you super experts can inspire me about a potential technical route to make it happen, and I can somehow help to setup and test that.

NiKiZe, 2021-09-15:
As a work around, have you considered using a HDMI splitter? 1 input, 2 output.
Connecting 2 keyboards to the machine should also work.

mdevaev, 2021-09-15:
Hi. We're already have issue for this case. Work in progress: #157

closed 2021-09-15 by mdevaev

Fisherworks, 2021-09-18:
> As a work around, have you considered using a HDMI splitter? 1 input, 2 output.
> Connecting 2 keyboards to the machine should also work.

well, that sounds like a reasonable and could-be-stable alternative before the #157 comes true.
Thanks and I'm gonna try this with a little more hardware stuff.

Fisherworks, 2021-09-18:
> Hi. We're already have issue for this case. Work in progress: #157

Thanks for the instant response, this's a super helpful and full-of-features project.