pikvm/pikvm #471: Guacamole connection How can I temporarily turn off the TLS of kvmd VNC function？

TelDragon, 2021-09-15:
How can I temporarily turn off the TLS of kvmd VNC function ？
请问如何临时关闭kvmd-vnc功能的tls？

I want to use guacamole to temporarily test the connection and find that the error log is TLS authentication error
我想使用guacamole 临时测试连接，发现错误日志为tls认证错误

It seems that not only guacamole does not support JPEG compression
好像不是 guacamole 不支持 JPEG 压缩导致。

Here is my test TTL log
以下是我的测试ttl日志

kvmd-vnc Console
````
[root@pikvm ~]# kvmd-vnc systemctl enable --now kvmd-vnc --run
kvmd.keyboard.keysym              INFO --- Reading keyboard layout /usr/share/kvmd/keymaps/en-us ...
kvmd.apps.vnc.server              INFO --- Listening VNC on TCP [::]:5900 ...
kvmd.apps.vnc.server              INFO --- [entry] [::ffff:192.168.1.18]:63194: Connected client
kvmd.apps.vnc.rfb                 INFO --- [entry] [::ffff:192.168.1.18]:63194: Starting client tasks ...
kvmd.apps.vnc.rfb                 INFO --- [main] [::ffff:192.168.1.18]:63194: Using RFB version 3.8
kvmd.apps.vnc.rfb                 INFO --- [main] [::ffff:192.168.1.18]:63194: Using VeNCrypt security type
kvmd.apps.vnc.rfb                 INFO --- [main] [::ffff:192.168.1.18]:63194: Using VeNCrypt/X509Plain auth type
kvmd.apps.vnc.rfb                ERROR --- [main] [::ffff:192.168.1.18]:63194: Unhandled exception
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 113, in __wrapper
    await coro
  File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 127, in __main_task_loop
    await self.__handshake_security()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 252, in __handshake_security
    await handler()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 304, in __handshake_security_vencrypt
    await self._start_tls(ssl_context, self.__tls_timeout)
  File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/stream.py", line 118, in _start_tls
    transport = await loop.start_tls(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1231, in start_tls
    await waiter
ConnectionResetError
kvmd.apps.vnc.rfb                 INFO --- [kvmd] [::ffff:192.168.1.18]:63194: Cancelling subtask ...
kvmd.apps.vnc.rfb                 INFO --- [streamer] [::ffff:192.168.1.18]:63194: Cancelling subtask ...
kvmd.apps.vnc.rfb                 INFO --- [fb_sender] [::ffff:192.168.1.18]:63194: Cancelling subtask ...
````

guacd Console
````
Sep 15 17:00:59 Privatization guacd[30251]: Creating new client for protocol "vnc"
Sep 15 17:00:59 Privatization guacd: guacd[30251]: INFO:#011Creating new client for protocol "vnc"
Sep 15 17:00:59 Privatization guacd: guacd[30251]: INFO:#011Connection ID is "$c51680ec-48b3-4038-95b8-606e40d7a059"
Sep 15 17:00:59 Privatization guacd[30251]: Connection ID is "$c51680ec-48b3-4038-95b8-606e40d7a059"
Sep 15 17:00:59 Privatization guacd[30757]: Cursor rendering: local
Sep 15 17:00:59 Privatization guacd: guacd[30757]: INFO:#011Cursor rendering: local
Sep 15 17:00:59 Privatization guacd: guacd[30757]: INFO:#011User "@53c1ed48-7ddd-4f70-8d7e-e36f99ad1300" joined connection "$c51680ec-48b3-4038-95b8-606e40d7a059" (1 users now present)
Sep 15 17:00:59 Privatization guacd[30757]: User "@53c1ed48-7ddd-4f70-8d7e-e36f99ad1300" joined connection "$c51680ec-48b3-4038-95b8-606e40d7a059" (1 users now present)
Sep 15 17:00:59 Privatization guacd[30757]: VNC server supports protocol version 3.8 (viewer 3.8)
Sep 15 17:00:59 Privatization guacd[30757]: We have 1 security types to read
Sep 15 17:00:59 Privatization guacd[30757]: 0) Received security type 19
Sep 15 17:00:59 Privatization guacd[30757]: Selecting security type 19 (0/1 in the list)
Sep 15 17:00:59 Privatization guacd[30757]: Selected Security Scheme 19
Sep 15 17:01:00 Privatization guacd[30757]: GnuTLS initialized.
Sep 15 17:01:00 Privatization guacd[30757]: Got VeNCrypt version 0.2 from server.
Sep 15 17:01:00 Privatization guacd[30757]: We have 3 security types to read
Sep 15 17:01:00 Privatization guacd[30757]: 0) Received security type 256
Sep 15 17:01:00 Privatization guacd[30757]: 1) Received security type 262
Sep 15 17:01:00 Privatization guacd[30757]: Selecting security type 262 (1/3 in the list)
Sep 15 17:01:00 Privatization guacd[30757]: 2) Received security type 259
Sep 15 17:01:00 Privatization guacd[30757]: Unsupported credential type requested.
Sep 15 17:01:00 Privatization guacd: guacd[30757]: ERROR:#011Unsupported credential type requested.
Sep 15 17:01:00 Privatization guacd[30757]: Reading credential failed
Sep 15 17:01:00 Privatization guacd[30757]: User "@53c1ed48-7ddd-4f70-8d7e-e36f99ad1300" disconnected (0 users remain)
Sep 15 17:01:00 Privatization guacd: guacd[30757]: INFO:#011User "@53c1ed48-7ddd-4f70-8d7e-e36f99ad1300" disconnected (0 users remain)
Sep 15 17:01:00 Privatization guacd: guacd[30757]: INFO:#011Last user of connection "$c51680ec-48b3-4038-95b8-606e40d7a059" disconnected
Sep 15 17:01:00 Privatization guacd[30757]: Last user of connection "$c51680ec-48b3-4038-95b8-606e40d7a059" disconnected
Sep 15 17:01:00 Privatization guacd[30251]: Connection "$c51680ec-48b3-4038-95b8-606e40d7a059" removed.
Sep 15 17:01:00 Privatization guacd: guacd[30251]: INFO:#011Connection "$c51680ec-48b3-4038-95b8-606e40d7a059" removed.
Sep 15 17:01:01 Privatization systemd: Started Session 1643 of user root.
Sep 15 17:01:01 Privatization systemd: Started Session 1641 of user root.
````

Please help me
I want to temporarily close VNC TLS to test the connection of guacamole.

请帮助我
我想临时关闭下vnc tls 测试guacamole 的连接。

arch1mede, 2021-09-15:
Please see [this](https://github.com/pikvm/pikvm/blob/e33dafecee3e70fb26b862fe027a90cc08c121a6/pages/vnc.md#unsupported-clients)

mdevaev, 2021-09-16:
To disable VNC TLS use this for `/etc/kvmd/override.yaml`:
```yaml
vnc:
    server:
        tls:
            ciphers: null
```

But Guacamole does not support JPEG compression, which is required for our VNC server. Given that it is now 2021, the lack of support for any compression (as well as encryption and authorization) in the VNC client is noncense. This problem should be fixed on the Guacamole side.

Alternatively, you can use TigerVNC - it works great.

closed 2021-09-16 by mdevaev

TelDragon, 2021-09-18:
Thank you very much for answering your questions and updating your code in time
非常感谢您的疑惑解答，以及您的代码及时更新