pikvm/pikvm #473: kvmd service does not start up after update & reboot

KarimGeiger, 2021-09-16:
**Describe the bug**
After updating to the latest version and rebooting, the kvmd service seems to no longer be working. As a result, the website returns with a 500 Internal Server Error.

I'm happy about any pointers as to how I can approach this issue. Re-flashing unfortunately isn't an option as I currently don't have physical access to the device.

**Expected behavior**
kvmd service should start properly.

**Log files**
```
journalctl:
Sep 16 16:53:09 kvm systemd[1]: Started Pi-KVM - The main daemon.
Sep 16 16:53:10 kvm kvmd[659]: Traceback (most recent call last):
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/bin/kvmd", line 9, in <module>
Sep 16 16:53:10 kvm kvmd[659]:     main()
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/kvmd/__init__.py", line 45, in main
Sep 16 16:53:10 kvm kvmd[659]:     config = init(
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/__init__.py", line 145, in init
Sep 16 16:53:10 kvm kvmd[659]:     config = _init_config(options.config_path, options.set_options, **load)
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/__init__.py", line 177, in _init_config
Sep 16 16:53:10 kvm kvmd[659]:     if _patch_dynamic(raw_config, config, scheme, **load_flags):
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/__init__.py", line 254, in _patch_dynamic
Sep 16 16:53:10 kvm kvmd[659]:     scheme["kvmd"]["auth"]["internal"].update(get_auth_service_class(config.kvmd.auth.internal.type).get_plugin_options())
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/auth/__init__.py", line 40, in get_auth_service_class
Sep 16 16:53:10 kvm kvmd[659]:     return get_plugin_class("auth", name)  # type: ignore
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/__init__.py", line 58, in get_plugin_class
Sep 16 16:53:10 kvm kvmd[659]:     module = importlib.import_module(f"kvmd.plugins.{sub}.{name}")
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
Sep 16 16:53:10 kvm kvmd[659]:     return _bootstrap._gcd_import(name[level:], package, level)
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap_external>", line 850, in exec_module
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/kvmd/plugins/auth/htpasswd.py", line 25, in <module>
Sep 16 16:53:10 kvm kvmd[659]:     import passlib.apache
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/passlib/apache.py", line 14, in <module>
Sep 16 16:53:10 kvm kvmd[659]:     from passlib.context import CryptContext
Sep 16 16:53:10 kvm kvmd[659]:   File "/usr/lib/python3.9/site-packages/passlib/context.py", line 17, in <module>
Sep 16 16:53:10 kvm kvmd[659]:     from passlib.utils import (handlers as uh, to_bytes,
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap_external>", line 846, in exec_module
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap_external>", line 978, in get_code
Sep 16 16:53:10 kvm kvmd[659]:   File "<frozen importlib._bootstrap_external>", line 647, in _compile_bytecode
Sep 16 16:53:10 kvm kvmd[659]: ValueError: bad marshal data (unknown type code)
Sep 16 16:53:10 kvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
Sep 16 16:53:11 kvm kvmd-cleanup[660]: kvmd.apps.cleanup                 INFO --- Cleaning up ...
Sep 16 16:53:11 kvm kvmd-cleanup[660]: kvmd.apps.cleanup                 INFO --- Trying to find and kill the streamer 'kvmd/streamer: <app>' ...
Sep 16 16:53:11 kvm kvmd-cleanup[660]: kvmd.apps.cleanup                 INFO --- Bye-bye
Sep 16 16:53:11 kvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
```

```
#systemctl status kvmd
* kvmd.service - Pi-KVM - The main daemon
     Loaded: loaded (/usr/lib/systemd/system/kvmd.service; enabled; vendor preset: disabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2021-09-16 16:55:21 MSK; 1s ago
    Process: 724 ExecStart=/usr/bin/kvmd --run (code=exited, status=1/FAILURE)
    Process: 725 ExecStopPost=/usr/bin/kvmd-cleanup --run (code=exited, status=0/SUCCESS)
   Main PID: 724 (code=exited, status=1/FAILURE)
```

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: Chrome
 - Version: Latest
 - VNC client: N/A

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version:
```
kvmd 3.23-1
kvmd-fan 0.1-1
kvmd-oled 0.3-1
kvmd-platform-v2-hdmi-rpi4 3.23-1
kvmd-webterm 0.40-1
```
  - uStreamer version: 4.5-1
  - Linux kernel: `Linux kvm 5.10.52-12-ARCH #1 SMP Wed Sep 1 05:47:56 MSK 2021 armv7l GNU/Linux`

mdevaev, 2021-09-17:
Looks like bad sd card

mdevaev, 2021-09-19:
Sup?

mdevaev, 2021-09-20:
No answer, so closing.

Let me know if the problem is still relevant.

closed 2021-09-20 by mdevaev