pikvm/pikvm #476: PS/2 passthrough support for Arduino HID firmware

mclei-asw, 2021-09-22:
I would like to support local console. It means VGA monitor and real PS/2 keyboard locally installed in the rack for local access in parallel to pikvm.

VGA could be easily split by Y cable. But it does not work for PS/2. So I would like to connect the real PS/2 or USB keyboard to Arduino with HID/PS2 firmware and let the Arduino to pass every key pressed on this keyboard to PS/2 output port connected to controlled computer.

This way it will be possible to remotely control the computer using pikvm console, but also locally using keyboard and LCD when present in the server room.

mdevaev, 2021-09-24:
It's a good idea, but I can't say that we will make it a priority.

mclei-asw, 2021-09-24:
I can sponsor somebody who is able to modify the Arduino firmware.

mdevaev, 2021-09-24:
Okay, I'll take that into account

No0ne, 2024-07-27:
With these two PRs this issue should be fixed/closed now:
https://github.com/pikvm/kvmd/pull/171
https://github.com/pikvm/kvmd/pull/175

mdevaev, 2024-07-27:
I'll write some docs and close this issue, yes.

No0ne, 2024-11-15:
Please adopt for docs.pikvm.org: https://github.com/No0ne/ps2x2pico?tab=readme-ov-file#ps2-passthru

mdevaev, 2024-11-15:
@No0ne Thank you! I will. I completely forgot about it :(

mdevaev, 2024-11-19:
Done.

closed 2024-11-19 by mdevaev