pikvm/pikvm #486: Create docker image if possible

benisai, 2021-09-26:
We can pass USB devices to docker, so a USB HDMI and USB host should be able to be passthrough to a container. This would be amazing if it can be done. We would all appreciate it! 💙🫂

arch1mede, 2021-09-26:
Please see [this](https://github.com/pikvm/pikvm/blob/master/pages/Community_FAQ.md#can-this-be-used-in-any-other-distros-like-rasbian-run-this-in-a-docker) and this #374

mdevaev, 2021-09-28:
Why you need this?

benisai, 2021-09-28:
> Why you need this?

Pikvm OS on a 4GB Pi is a waste of resource. If it was a container, we could run it along other services.

mdevaev, 2021-09-28:
PiKVM is a means of emergency access to the server. It comes in the form of an image that is already configured in an optimal way to ensure reliability.

1. PiKVM cannot be placed in one container because it consists of many services.
2. For the correct operation of PiKVM, special OS settings (sysctl, etc.) are needed.
3. The PiKVM runs on a read-only file system to eliminate crashes due to sudden power loss.

Please don't try to cram something into the same Raspberry that does not relate to the functions of PiKVM. You'll shoot yourself in the foot. All this can lead to the fact that at the moment when you need PiKVM the most, it will not work, because some service has corrupted its memory card or something like that.

mdevaev, 2021-09-28:
In general, since this is contrary to the ideology of the project, I will not do it myself. But I am ready to accept patches that simplify the use in the docker (if they do not break the current behavior of the system). After all, this is an open source, you are free to use it at your discretion.

closed 2021-09-28 by mdevaev

mdevaev, 2025-09-25:
Now it's impossible due a huge amount of kernel patches and tunings of operating system.