pikvm/pikvm #505: Feature - Removing RPi specific dependencies from kvmd

shantur, 2021-10-13:
Bringing this from the discussion over Discord and tracking the required tasks here.

The idea is to make kvmd platform independent, currently it is dependent on RPi specific stuff which gets installed with kvmd when trying to install it on another arm platform running arch linux ( like RockPro64 )

Below are the ones which are left from my findings.

- [x] raspberrypi-firmware
- [ ] raspberrypi-io-access
- [ ] ustreamer with OMX
- [ ] https://github.com/pikvm/kvmd/blob/master/scripts/kvmd-udev-hdmiusb-check

mdevaev, 2021-11-07:
Moved `kvmd-udev-hdmiusb-check` to `kvmd-platform`

mdevaev, 2021-11-08:
`ustreamer` can't be removed from dependencies, but you can build your own package like `ustreamer-generic` with `provides` and `conflicts` for it. Example: https://github.com/pikvm/packages/blob/master/packages/linux-api-headers-pikvm/PKGBUILD#L20

mdevaev, 2021-11-21:
UPD: OMX and MMAL will be removed from streamer altogether and replaced with a V4L2 driver.

mdevaev, 2022-03-26:
raspberrypi-io-access has been replaced by abstract platform-io-access

closed 2022-03-26 by mdevaev