pikvm/pikvm #523: Pi Zero start reboot when I open KVM.

alexsalex, 2021-10-30:
Will provide any additional info by request.

```
Oct 30 04:02:44 ms-kvm kvmd[440]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://ms-kvm.local/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.1 Safari/605.1.15'
Oct 30 04:02:44 ms-kvm kvmd[440]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://ms-kvm.local/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.1 Safari/605.1.15'
Oct 30 04:02:44 ms-kvm kvmd[440]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://ms-kvm.local/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.1 Safari/605.1.15'
Oct 30 04:02:44 ms-kvm kvmd[440]: aiohttp.access                    INFO --- [admin (token) / 10.0.1.125] 'GET /auth/check HTTP/1.0' => 200; size=199 --- referer='https://ms-kvm.local/kvm/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.1 Safari/605.1.15'
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.server             INFO --- Registered new client socket: WsClient(id=1947773584, stream=True); clients now: 1
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- Starting streamer ...
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=774: ['/usr/bin/ustreamer', '--device=/dev/kvmd-video', '--persistent', '--dv-timings', '--format=uyvy', '--encoder=omx', '--workers=1', '--quality=50', '--desired-fps=30', '--drop-same-frames=30', '--last-as-blank=0', '--unix=/run/kvmd/ustreamer.sock', '--unix-rm', '--unix-mode=0660', '--exit-on-parent-death', '--process-name-prefix=kvmd/streamer', '--notify-parent', '--no-log-colors']
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.201      main] -- Using internal blank placeholder
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.220      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.221      http] -- Starting HTTP eventloop ...
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.221    stream] -- Using V4L2 device: /dev/kvmd-video
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.221    stream] -- Using desired FPS: 30
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.225    stream] -- Device fd=9 opened
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.225    stream] -- Using input channel: 0
Oct 30 04:02:45 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.242    stream] -- Got new DV timings: resolution=1280x720, pixclk=74250000
Oct 30 04:02:46 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.322    stream] -- Using resolution: 1280x720
Oct 30 04:02:46 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.323    stream] -- Using pixelformat: UYVY
Oct 30 04:02:46 ms-kvm kvmd[440]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [352.323    stream] -- Querying HW FPS changing is not supported
client_loop: send disconnect: Broken pipe
```

mdevaev, 2021-10-30:
Try to change the memory card, it may be broken

arch1mede, 2021-10-30:
I am unable to repo this...please join discord to see if we can assist.

mdevaev, 2021-10-31:
@alexsalex sup?

mdevaev, 2021-11-01:
There is no response from the author, so I'm closing it.

If you still want to deal with this issue, please contact our technical support at Discord.

closed 2021-11-01 by mdevaev

alexsalex, 2021-11-01:
Sorry, was on vocation.

Could you tell me how can I find you in discord?

arch1mede, 2021-11-01:
www.pikvm.org, at the top, there is a discord button.

alexsalex, 2021-12-03:
So, I tried to plug the Pi Zero in to power supply, and do not connect usb to Pi. It works fine, I can see screen, but can't control it.

When I connect usb, got rebooted.

[image]

alexsalex, 2021-12-03:
Could you open the issue?

Another thing, when the Pi booting, I can see the long boot for PiKVM daemon and some related daemons. Will check the logs...

alexsalex, 2021-12-03:
[image]

alexsalex, 2021-12-03:
Last one power from power supply, usb connected to server.

mdevaev, 2021-12-03:
Please perform a reflash. You are using some kind of unstable kernel. I don't know where you got it, but it's not what the latest version is.

alexsalex, 2021-12-03:
Regular update (have ansible playbook). Could you tell me how to resolve this issue?

[image]

mdevaev, 2021-12-04:
You have a very old OS with the wrong kernel packages. Please do a reflash, it will be easier.

alexsalex, 2021-12-04:
Hm... fixed by fresh install.

But guys, I'm not invent the pikvm and do not use Arch Linux at all (rpm/deb guy). I got this image from you, and regularly updated it. How can I prevent this situation in the future?

mdevaev, 2021-12-04:
In the future, this will not happen again because the new images are arranged a little differently than the old ones. You had a VERY old image before the kernel package changes.

alexsalex, 2021-12-04:
Thank you for your help! Thank you for this software and hardware) It's amazing.

Btw, fresh install fix issue with iPad, now it works, but not in Safari (Firefox and Chrome works fine).

mdevaev, 2021-12-06:
> but not in Safari (Firefox and Chrome works fine).

What do you mean, which issue?

alexsalex, 2021-12-06:
On iPad Safari I can't login... even if I can, when I open the KVM window I can see the connection icon is blinking orange forever.

The same issue if I use nginx proxy. KVM can't connect. But I think I will ask this in new post.

mdevaev, 2021-12-06:
This is an iOS bug. You need to either disable HTTPS or install a valid SSL certificate. io WebSocket does not work with self-signed certificates. For more details, contact the Discord support chat,

alexsalex, 2021-12-06:
Looks like it's not iOS issue, but Safari on iOS.

[image]

mdevaev, 2021-12-07:
Oh, at least you can use it without Safari