pikvm/pikvm #524: I²C USB HID

kmeaw, 2021-10-31:
**Is your feature request related to a problem? Please describe.**
In order to emulate keyboard and mouse, you need to connect some USB device to the PC.
It can be either Pi's usb-gadget or some other device with both a USB interface and another interface to the Pi.
Arduino seems to be a good target - it is well known and popular enough to have a good chance that a maker already has one.
PiKVM suggests using Arduino Pro Micro connected to the Pi through SPI.
However, Arduino uses 5V TTL levels opposed to the 3.3V for the Pi, so you have to use level shifters to avoid damaging the SoC.

**Describe the solution you'd like**
Switching from SPI to I²C can solve this problem - Pi would pull the data line high (to 3.3V) and Arduino would pull it down (to GND). The protocol guarantees that the line would not be pulled up to 5V so it is safe to interconnect without a level shifter.
I would like to have an I²C firmware for the Arduino and a PiKVM counterpart which would stream HID events over I²C.

**Describe alternatives you've considered**
I have tried flashing a Blue Pill with https://github.com/microxblue/pikvm_hid but for some reason the device would not be recognized by the PC as a valid USB HID device.

**Additional context**
Also using I²C instead of SPI would allow connecting multiple Arduinos to the same bus. With an HDMI switcher it would allow a single PiKVM to control multiple servers. SPI would require at least a separate CS line to each Arduino.

mdevaev, 2021-10-31:
SPI has a significant advantage and allows you to perform firmware directly from pikvm. In general, the idea is interesting, but so far I can't say that this is a priority direction. If you investigate this and make a patch, I will accept it.

Example of the phy layer for spi: https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/hid/spi.py

The firmware: https://github.com/pikvm/kvmd/tree/master/hid

NiKiZe, 2021-10-31:
You can also run the Pro Micro on 3.3V, lower the internal frequency, disconnect 5v from usb, and instead power it from the Pi?

mdevaev, 2023-08-08:
@kmeaw @NiKiZe We have some movement on this issue. I made a new firmware for Pico/RP2040, which solves all the problems at once:
* The RP2040 runs on 3.3V, so there is no need for a level shifter.
* SPI is used instead of UART, so from the Raspberry Pi side, the UART will be free.
Using Arduino will be marked as legacy. The old devices and code will of course continue to work, but for new devices I recommend Pico. This will be called PiKVM V1, and now an instruction is ready that explains how to connect the Pico HID to V2 or V3: https://docs.pikvm.org/pico_hid/

mdevaev, 2023-08-11:
I've documented about replacement Arduino to Pico: https://docs.pikvm.org/pico_hid/#arduino-hid-replacement

I believe that the problem can be considered settled with the Pico HID.

closed 2023-08-11 by mdevaev