pikvm/pikvm #527: Why can't we have HDMI and USB captures from both target systems?

ramesharige25, 2021-11-01:
Scenario.
1) I have HDMI 2 CSI cable that capture stream from Target PC1 
2) USB dongle to capture stream from Target PC2. 

Can't we enable both of them together with new piKVM build (if possible) and control both machines?

As of now, rpi4 has only one data and power input, can't we modify the rpi4 to have two data inputs with some hats or board and enable single pi to capture KVM's from two target pc's ??

NiKiZe, 2021-11-01:
You should probably use KVM switch to have that ability

ramesharige25, 2021-11-01:
I agree that, using switch you are able access one of them at a time. But if we can implement this, we can access 2 systems at same time in the same browser like below.

https://pikvm/1
https://pikvm/2.

I am not sure if possible or not, but we can do one thing for sure, we can do KVM with Target 1 PC using HDMI to CSI and Target 2 PC we can at least capture video stream using USB dongle, but need to find a way to control Keyboard and Mouse. I feel there should be way for this for sure.

mdevaev, 2021-11-01:
It is possible to use several video capture devices, but pikvm can control only one, the second one will have to be configured manually. At least that's the case now.

But it is fundamentally impossible to make several keyboard and mouse emulation channels, since the Pi 4 has only one free USB controller.

closed 2021-11-01 by mdevaev