pikvm/pikvm #531: Fullscreen Video still shows resize border when focused

Nickno95, 2021-11-03:
**Describe the bug**
When the video window is fullscreened and in focus, the blue resize border appears, shrinking the overall video slighly and making picture quality blurrier.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to KVM
2. Click on the fullscreen arrows in the video window
3. Clicking in the window will show the border which shrinks the video size slightly
4. Clicking outside like in a different program on another monitor will make the border disappear

**Expected behavior**
The resize border should be hidden when fullscreened, you don't resize fullscreened windows and it takes up unnecessary space

**Screenshots**
When not in focus (without resize border):
[image]

When focused (with resize border):
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 95

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.40-1
  - uStreamer version: 4.9-1
  - Linux kernel: Linux pikvm 5.10.52-12-ARCH # 1 SMP Wed Sep 1 05:47:56 MSK 2021 armv7l GNU/Linux

mdevaev, 2021-11-03:
This is not a bug, but a feature. The browser window may lose focus, so the blue frame shows that everything is working as it should.

Nickno95, 2021-11-03:
Could there be a toggle option for this? a 1080p stream on a 1080p monitor looks terrible the moment the window is in focus and makes some text harder to read, but is crisp otherwise. maybe even an alternate small watermark in a corner, something that doesn't resize the stream.

mdevaev, 2021-11-03:
Okay

mdevaev, 2021-11-04:
Done. No more resizes. Please update and have fun

closed 2021-11-04 by mdevaev

eychei, 2024-09-12:
Hi,

I have the same issue within windowed mode. Can you also remove the resizing there please?
I am using windowed mode as a app function in a chrome tab. So when ever I focus on the tab I get the blue border which then resizes the frame just a few pixels and thus make the image blurry.
Hope there is a solution for that.

Thanks.

Ey

eychei, 2024-09-12:
Ok I found the solution:
Change web.css to remove Fullscreen boarder and Windowed boarder:

:root {
--shadow-window-fullscreen-active: 0 0 0 0px #5b90bb00 inset !important;
--border-window-active-2px: 0px solid #5b90bb;
}

Tycho-S, 2025-01-06:
I still also get this with the latest PiKVM 0.25.1 . I thought this was removed in fullscreen mode in 2021? Maybe it was put back?