pikvm/pikvm #533: Support Raspberry pi Zero 2 W

ogerd, 2021-11-04:
Raspberry introduce new PI Zero 2 W model. There are better performance. It does make sense to upgrade to Pi Zero 2 (RAM still on 512MB)?

Please could you plan to support Raspberry Pi Zero 2 W? I try it on, but there is problem with wifi (device not recognized).

Thanks

arch1mede, 2021-11-04:
We are aware, WIP

coolinger, 2021-11-05:
Hi, I just got it working, here are my steps:

To get wifi working, you need to build a new image with new kernel, and additionally copy the drivers for wifi, described here:
https://archlinuxarm.org/forum/viewtopic.php?f=9&t=15633&sid=37deda5505050afd541f980fa7f8a595&start=10

The CSI Interface was not working out of the box. The kernel name changed to 3f801000.csi. I had to extend a line in /etc/udev/rules.d/99-kvmd.rules:
KERNEL=="video[0-9]*", SUBSYSTEM=="video4linux", KERNELS=="20801000.csi|20801000.csi1|3f801000.csi", ATTR{name}=="unicam-image", GROUP="kvmd", SYMLINK+="kvmd-video", TAG+="systemd"

mdevaev, 2021-11-05:
The new kernel is broken:
* https://github.com/raspberrypi/linux/issues/4669
* https://github.com/raspberrypi/linux/issues/4670

We need to fix this before.

pete1019, 2021-11-06:
@coolinger how is the performance? How many FPS? How does the latency feel?
Thx

arch1mede, 2021-11-06:
> @coolinger how is the performance? How many FPS? How does the latency feel?
> Thx

This is not the appropriate venue, this is still a wip, please take this to discord.

coolinger, 2021-11-06:
FPS around 10-20, this is much better then before. But the latency is still the same, around 2 seconds. With MJPEG 1280x720 and gdm login manager running.

On console I have around 10 fps, and nearly no latency (but I did not try this with the old zerow)

arch1mede, 2021-11-06:
> FPS around 10-20, this is much better then before. But the latency is still the same, around 2 seconds. With MJPEG 1280x720 and gdm login manager running.
> 
> On console I have around 10 fps, and nearly no latency (but I did not try this with the old zerow)

Please stop posting stats, this issue ticket is about getting pikvm working on it, this is still too early in development to start posting performance.

pmagid, 2021-11-12:
+1 eagerly awaiting support

ChrSch84, 2021-11-14:
+1 eagerly awaiting support...here too!

chilman408, 2021-11-19:
hello,

is there a beta image for the Pi Zero 2 to test out?  Happy to provide feedback on it.

mdevaev, 2021-11-20:
Here the test image: https://files.pikvm.org/images/v2-hdmi-zero2w-latest.img.xz

Please try this and let me know if everything is OK. The instructions are exactly the same as in the case of the regular zero w.

mdevaev, 2021-11-20:
UPD: download it again from the same link, fixed some.

luizribeiro, 2021-11-20:
That worked perfectly for me, thanks for sharing the WIP image @mdevaev!

pete1019, 2021-11-20:
@mdevaev are many USB-Ethernet supported out of the box?

For example this has a **RTL8152B**
https://www.waveshare.com/wiki/ETH/USB_HUB_HAT_(B)
or here as well
https://www.berrybase.de/en/raspberry-pi/raspberry-pi-computer/usb-devices/micro-usb-2.0-fast-ethernet-netzwerkkonverter-wei-223?c=291

Thanks a lot!

mdevaev, 2021-11-20:
@pete1019 You will not be able to use USB ethernet together with USB emulation, since PiZero has only one controller. The second Micro USB connector can only be used for power

mdevaev, 2021-11-20:
@luizribeiro Great! If this image works, it means it is no longer WIP :D

pete1019, 2021-11-20:
@mdevaev 
They use a USB Hub and you will have multiple USB Ports then:
https://www.waveshare.com/wiki/ETH/USB_HUB_HAT_(B)

Will this work?
Thanks

mdevaev, 2021-11-20:
No. PiZero (any) has **only one USB controller**. It can be either a gadget or a host. These are exclusive modes.

ChrSch84, 2021-11-20:
@mdevaev Thanks for the Image. On my Pi Zero 2 it is working. But i have little Bug. It shows me the H264 Button but if i use it it does not work. Is it supported?

mdevaev, 2021-11-20:
Yes. If H264 does not work, it is probably due to network problems. See here: https://docs.pikvm.org/webrtc/#troubleshooting

If nothing helps, please contact the Discord #support chat. The problem is definitely not specific to Zero.

mdevaev, 2021-11-20:
Okay, it's done

closed 2021-11-20 by mdevaev