pikvm/pikvm #542: Move environment from config.mk to environment file

bbeaudoin, 2021-11-20:
At present, `config.mk` has options for 

Suggestion for image build (config.mk parsing issues into Makefile for docker run). There is an --env-file that can be used with docker rather than -e or --env so configuration can be placed into the environment file rather than parsed by make.

Both -e and --env-file can be passed in the same line: https://docs.docker.com/engine/reference/commandline/run/#set-environment-variables--e---env---env-file

In theory this would simpliffy the `docker` commandline by reducing the number of `-e` or `--env` flags required and eliminate the possibility special characters in passphrases would be interpolated by bash when `docker run` is invoked by `make`.

mdevaev, 2021-11-29:
Maybe it's worth doing

mdevaev, 2022-05-02:
I just read about this feature. Environment variables cannot be used at the `docker build` stage. Sorry.

closed 2022-05-02 by mdevaev

bbeaudoin, 2022-05-02:
Technically true, it's still possible in a few ways (https://vsupalov.com/docker-build-pass-environment-variables/). It's also possible to source the environment variables using the Makefile itself (default is to use /bin/sh, that could be changed to /bin/bash to source the environment file) to avoid the extra `make` processing of variables:

```make
# Makefile

SHELL := /bin/bash

rule:
    source env.sh && YourCommand
```

Ultimately there are more configuration options (i.e. `pikvm.txt`) which means less need to rely on compile-time options for these settings so accepting things like wireless passwords at compile time is no longer a big issue.

mdevaev, 2022-05-02:
I don't like the shell sourcing. I think we can add more options to pikvm.txt.