pikvm/pikvm #547: XH-HK4401 HDMI Switch drivers

shiecldk, 2021-11-26:
Hi there, I read the guide for XH-HK4401 in here:
https://docs.pikvm.org/xh_hk4401/

It mentions using the onboard UART pin connection without an USB UART. However, in the "Adding UI elements to control the KVM switch" section, it does not state what dirvers I should use for the onboard UART.

I've tried changing the line "device: /dev/ttyUSB0" to "device: /dev/ttyAMA0", assuming this would use the onboard UART. However, it does not work. The inputs light in the GPIO menu is still grey out and I cannot switch the GPIO with the CLICK button either (also grey out).

Please provide me help on this. Thank you very much.

shiecldk, 2021-11-26:
@ep1cman @mdevaev Do you know how to solve this?

mdevaev, 2021-11-26:
Support was happening in the chat.

closed 2021-11-26 by mdevaev