pikvm/pikvm #552: 1  "RuntimeError: Can't find any UDC" on RPI4B

makefu, 2021-11-28:
**Describe the bug**
i tried to get PiKVM running on my RPI4B with the C779 CSI bridge. However, when starting up and trying to connect to the webserver i get 500 server error and the kvmd console shows `RuntimeError: Can't find any UCD`

**To Reproduce**
Steps to reproduce the behavior, like:
1. Download and flash latest pikvm image onto an sdcard
2. boot rpi4b
3. browse to https://pikvm/
4. See error 500

**Expected behavior**
Application server starts up

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:Linux
 - Browser: Firefox

**PiKVM info:**
  - Raspberry Pi board version RPi 4B
  - PiKVM platform: v2-hdmi
  - Video capture type: CSI bridge C779
  - KVMD version: `pacman -Q | grep kvmd`:
```
kvmd 3.49-1
kvmd-fan 0.1-1
kvmd-oled 0.8-1
kvmd-platform-v2-hdmi-rpi4 3.49-1
kvmd-webterm 0.41-1
```
  - uStreamer version: `pacman -Q | grep ustreamer`: `ustreamer 4.9-1`
 
  - Linux kernel: `uname -a`: ` Linux pikvm 5.10.52-13-ARCH #1 SMP Sat Nov 20 09:40:57 MSK 2021 armv7l GNU/Linux`

**Additional context**

Related: https://old.reddit.com/r/pikvm/comments/r44zro/runtimeerror_cant_find_any_udc_on_rpi4b/

* journalctl -xu kvmd: https://p.krebsco.de/03l21c0
* journalctl -k: https://p.krebsco.de/1h7n346

mdevaev, 2021-11-28:
Please show output of the `pistat` command.

makefu, 2021-11-29:
```
[root@pikvm ~]# pistat
# Raspberry Pi 4 Model B Rev 1.1

CPU temp: 27.74'C

```
After CPU the program hangs on pikvm.
I booted Raspbian bullseye, copied over the script and ran it:
```
root@raspberrypi:~# bash ./lol
# Raspberry Pi 4 Model B Rev 1.1

CPU temp: 32.61'C
GPU temp: 32.1'C

Throttled flags: 0x0

Throttled now:  no
Throttled past: no

Undervoltage now:  no
Undervoltage past: no

Frequency capped now:  no
Frequency capped past: no
```

It seems that the vcgencmd is hanging on pikvm already

mdevaev, 2021-11-29:
Please contact me using discord PM.

makefu, 2021-11-29:
@mdevaev will do!

makefu, 2021-11-29:
@mdevaev thanks for the debugging session, I will close this issue for now, it may be related to Hardware, the board is RPI4 with Revision 1.1

rafuz, 2021-11-30:
I have the same problem on the exactly same hardware.
Has it been fixed?

potatoattack, 2021-11-30:
I'm also having this same issue after updating 10 minutes ago (hadn't updated for a few weeks).

mdevaev, 2021-11-30:
Do you have 1.1 revision of 4Gb board to?

mdevaev, 2021-11-30:
Apparently, the issue is the incompatibility of stock firmware with the old board revision 1.1.

Please try older images and tell me which one will work fine: https://files.pikvm.org/images/

makefu, 2021-12-01:
I tested the last Image ( https://files.pikvm.org/images/v2-hdmi-rpi4-20211022.img.xz ) and the error also occurred.

HardwoodHomelab, 2021-12-08:
I'm having this same issue with a Rev. 1.2 board 4gb board.

arch1mede, 2021-12-09:
> I'm having this same issue with a Rev. 1.2 board 4gb board.

Please join discord for this issue, I have 1.2 boards and they all work.

mdevaev, 2021-12-12:
I have a suspicion that there may be an old EEPROM on your boards. Or vice versa too new. Please run:
```
pacman -Syu
pacman -S rpi-eeprom
rpi-eeprom-update
```
And post results of the last command here.

HardwoodHomelab, 2021-12-12:
I ran those commands and the output of rpi-eeprom-update is:

*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
    LATEST: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
   RELEASE: critical (/lib/firmware/raspberrypi/bootloader/critical)

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

mdevaev, 2021-12-12:
Try to update `rpi-eeprom-update -a`

HardwoodHomelab, 2021-12-12:
I updated the eeprom and rebooted the PI. The rpi-eeprom-update output now shows:

BOOTLOADER: up to date
CURRENT: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
LATEST: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
RELEASE: critical (/lib/firmware/raspberrypi/bootloader/critical)

VL805_FW: Dedicated VL805 EEPROM
VL805: up to date
CURRENT: 000138a1
LATEST: 000138a1

I am still getting the can't find UDC error.

mdevaev, 2021-12-12:
Try to append line `core_freq_min=250` to `/boot/config.txt`

mdevaev, 2021-12-12:
If it will not work, try following:
```
sed -i -e s/critical/stable/g /etc/default/rpi-eeprom-update
rpi-eeprom-update -a
```

arch1mede, 2021-12-13:
HardwoodHomelab's issue was not the same, I had him reflash and he reported that everything was working normally.

mdevaev, 2021-12-13:
@makefu @rafuz try reflashing on a new memory card

mdevaev, 2021-12-16:
@makefu @rafuz sup?

arch1mede, 2021-12-16:
Adding to this convo, I know someone that has a 1.1 board and after flashing, working fine

mdevaev, 2021-12-16:
Flashing EEPROM or OS?

arch1mede, 2021-12-16:
> Flashing EEPROM or OS?

OS

mdevaev, 2021-12-16:
Okay, then we believe that flashing and using another memory card helps with this.
Users, please let us know if this is not the case.

makefu, 2021-12-17:
i've got another RPI 1.4 and used the same sd card (without reflashing). The issue didn't occur.

Martin-HiPi, 2021-12-18:
The EEPROM is stored in a chip on the Pi4, so swapping the card isn't good for testing this. 
Please run `rpi-eeprom-update` and post the result here please.

To update, 
```
pacman -Syu
pacman -S rpi-eeprom
rpi-eeprom-update -a
```

Martin

makefu, 2021-12-19:
actually i ran rpi-eeprom-update when i had the 1.1 version with the same issue occurring. however due to my swap i currently do not have access to the board

mdevaev, 2021-12-19:
Soon I will make an experimental release of the kernel and we will see if there is still this problem

DrMarlin, 2022-01-08:
I have this same issue - does anyone happen to know the eeprom version that works to force a downgrade?

mdevaev, 2022-01-08:
@DrMarlin try another memory card and full reflash.

DrMarlin, 2022-01-09:
@mdevaev That did it - thanks!

mdevaev, 2022-02-05:
I'm closing it because apparently the problem is related to the memory cards

closed 2022-02-05 by mdevaev