pikvm/pikvm #553: Keyboard and mouse not working?

amod-amatya, 2021-12-02:
I just read the FAQ section for 
USB problems (keyboard, mouse, mass storage, etc) [https://docs.pikvm.org/faq/#usb-problems-keyboard-mouse-mass-storage-etc](url)

It has been mentioned that a usb hub wont work and use Y splitter instead. Why is that so?
I am currently using this 3 in 1 usb hub from uni [shorturl.at/ouvV6](url)

Looking forward for your response.

mdevaev, 2021-12-02:
Because the core USB controller can work either in host mode or in gadget mode. The hub can only work with the tail.

For support questions, please contact Discord.

closed 2021-12-02 by mdevaev

amod-amatya, 2021-12-02:
Do you think this device should do the job?
I already have usb c to usb a cable
https://www.amazon.de/RIIEYOCA-Splitter-Adapter-Control-Interface/dp/B098PYB42L/ref=asc_df_B098PYB42L/?tag=googshopde-21&linkCode=df0&hvadid=526284679152&hvpos=&hvnetw=g&hvrand=5839684103057302153&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9042253&hvtargid=pla-1403827393163&psc=1&th=1&psc=1

mdevaev, 2021-12-03:
I don't know what circuit is inside this cable. you really should ask this in discord, because there may be people who use (or have tried) a similar cable.