pikvm/pikvm #554: PiKVM Hat v3 Oled screen issue

kitzler-walli, 2021-12-03:
I just received my PiKVM hat v3 by mail and after assembling everthing seems to work just fine except for the oled screen. Do I have to enable it?
I used the os image provided specifically for this hardware so I thought that should work out of the box.

mdevaev, 2021-12-04:
Hello. The OLED must be turned on after manual assembly, just like the fan controller. Pay attention to point 5 in the user manual: 
[image]

For support questions, you should contact [Discord](https://discord.gg/bpmXfz5) chat, not Github. There you will get an answer much faster from our specialists, and here you will have to wait until I wake up, because for sure we live in different time zones :)

closed 2021-12-06 by mdevaev

kitzler-walli, 2021-12-09:
Thank you for your support.