pikvm/pikvm #559: No Wifi through on pi

antidefran, 2021-12-08:
**Describe the bug**
I haven't been able to get my PiKVM onto wifi reliably

**To Reproduce**
I've tried to set up wifi through the pikvm.txt but the change never seems to take effect. I've also tried the wpa_passphrase command and also through netctl profiles. After reboot, the wpa_supplicant-wlan0.config changes back to empty. I've made sure to mount the file system to rw and back to ro when I'm finished.

**Expected behavior**
I've tried to connect a the pikvm to 2 different wifi's: the first is a standard wpa2 network and the other is an open network that uses Mac authentication. neither of these networks work and the changes never seem to take effect. 

**PiKVM info:**
  - rpi-4
  - PiKVM hat v3  
 - csi bridge
- kvmd 3.48-1
-kvmd-fan 0.1-1
-kvmd-oled 0.8-1
-kvmd-platform-v3-hdmi-rpi4 3.48-1
-kvmd-webterm 0.41-1
  - ustreamer 4.9-1
  - Linux kernel: Linux pikvm 5.10.52-13-ARCH #1 SMP Sat Nov 20 09:40:57 MSK 2021 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-12-08:
Hello. Please don't use netctl, this is a legacy way to configure network.
Try to follow this guide:
https://docs.pikvm.org/wifi

arch1mede, 2021-12-08:
Please join discord for troubleshooting

mdevaev, 2021-12-08:
Solved in discord.

closed 2021-12-08 by mdevaev