pikvm/pikvm #561: cannot update pikvm

connected201, 2021-12-09:
Hi, i cannot execute succesfully command `pacman -Syu`

```
[root@pikvm kvmd-webterm]# rw
[root@pikvm kvmd-webterm]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
:: Synchronizing package databases...
 core is up to date
 extra                1644.7 KiB   199   B/s 57:21 [################-------]  71%
 extra.db failed to download
error: failed retrieving file 'extra.db' from de3.mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to synchronize all databases (download library error)
[root@pikvm kvmd-webterm]# 
```
Dwownloadin crashed every time on this step (`1644.7 KiB   199   B/s 57:21 [################-------]  71%`)
Raspberry connected to network with cable.

connected201, 2021-12-09:
Update: Strange things, in Lan network, ping to raspberry is average 364ms, but my laptop is connectet do my network with lan cable, i will investigate problem

closed 2021-12-09 by connected201

fin444, 2024-10-14:
I had this issue also, fixed with reboot.