pikvm/pikvm #564: kvmd-janus :No Janus API transport is available... enable at least one and restart Janus

TelDragon, 2021-12-12:
errorinfo:
```
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => [WARN] Data Channels support not compiled
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Event handlers support disabled
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Plugins folder: /usr/lib/ustreamer/janus
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Joining Janus requests handler thread
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Loading plugin 'libjanus_ustreamer.so'...
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Sessions watchdog started
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Initializing plugin ...
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Reading config file '/etc/kvmd/janus/janus.plugin.ustreamer.jcfg' ...
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => [WARN] The 'janus.plugin.ustreamer' plugin doesn't implement any callback for RTP/RTCP/data... is this on purpose?
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Transport plugins folder: /usr/lib/janus/transports
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => Loading transport plugin 'libjanus_websockets.so'...
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected
Dec 12 18:06:55 j3r-k kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => [ERR] [janus.c:main:5495] #011Couldn't load transport plugin 'libjanus_websockets.so': /usr/lib/janus/transports/libjanus_websockets.so: undefined symbol: lws_validity_confirmed
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => [FATAL] [janus.c:main:5557] No Janus API transport is available... enable at least one and restart Janus
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner           ERROR --- Unexpected Janus error: pid=1945
Dec 12 18:06:55 pi kvmd-janus[500]: Traceback (most recent call last):
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/local/lib/python3.7/dist-packages/kvmd-3.50-py3.7.egg/kvmd/apps/janus/runner.py", line 156, in __janus_task_loop
Dec 12 18:06:55 pi kvmd-janus[500]:     raise RuntimeError("Janus unexpectedly died")
Dec 12 18:06:55 pi kvmd-janus[500]: RuntimeError: Janus unexpectedly died
Dec 12 18:06:55 pi kvmd-janus[500]: kvmd.apps.janus.runner           ERROR --- Can't kill process pid=1945
Dec 12 18:06:55 pi kvmd-janus[500]: Traceback (most recent call last):
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/local/lib/python3.7/dist-packages/kvmd-3.50-py3.7.egg/kvmd/apps/janus/runner.py", line 156, in __janus_task_loop
Dec 12 18:06:55 pi kvmd-janus[500]:     raise RuntimeError("Janus unexpectedly died")
Dec 12 18:06:55 pi kvmd-janus[500]: RuntimeError: Janus unexpectedly died
Dec 12 18:06:55 pi kvmd-janus[500]: During handling of the above exception, another exception occurred:
Dec 12 18:06:55 pi kvmd-janus[500]: Traceback (most recent call last):
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/local/lib/python3.7/dist-packages/kvmd-3.50-py3.7.egg/kvmd/aioproc.py", line 99, in kill_process
Dec 12 18:06:55 pi kvmd-janus[500]:     proc.terminate()
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/lib/python3.7/asyncio/subprocess.py", line 131, in terminate
Dec 12 18:06:55 pi kvmd-janus[500]:     self._transport.terminate()
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/lib/python3.7/asyncio/base_subprocess.py", line 151, in terminate
Dec 12 18:06:55 pi kvmd-janus[500]:     self._proc.terminate()
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/lib/python3.7/subprocess.py", line 1751, in terminate
Dec 12 18:06:55 pi kvmd-janus[500]:     self.send_signal(signal.SIGTERM)
Dec 12 18:06:55 pi kvmd-janus[500]:   File "/usr/lib/python3.7/subprocess.py", line 1746, in send_signal
Dec 12 18:06:55 pi kvmd-janus[500]:     os.kill(self.pid, sig)
Dec 12 18:06:55 pi kvmd-janus[500]: ProcessLookupError: [Errno 3] No such process
```
I am trying to migrate this project to other Debian systems
我正尝试将该项目移植到其他debian系统中

Janus installation steps
```
wget https://github.com/meetecho/janus-gateway/archive/refs/heads/master.zip
wget https://webrtc.github.io/adapter/adapter-latest.js

./autogen.sh
./configure \
--prefix=/usr \
--sysconfdir=/etc \
--disable-docs \
--disable-data-channels \
--disable-turn-rest-api \
--disable-all-plugins \
--disable-all-loggers \
--disable-all-transports \
--enable-websockets \
--disable-sample-event-handler \
--disable-websockets-event-handler \
--disable-gelf-event-handler

make -j4 && make -j4 install

sed -i -e 's|^#include "refcount.h"$|#include "../refcount.h"|g' "/usr/include/janus/plugins/plugin.h"
sed -i -e 's|^function Janus(|export function Janus(|g' "/usr/share/janus/javascript/janus.js"
sed -i '1s|^|import "./adapter.js"\n|' "/usr/share/janus/javascript/janus.js"
cp ../adapter-latest.js "/usr/share/janus/javascript/adapter.js"

janus -V
Janus commit: not-a-git-repo
Compiled on:  Sun 12 Dec 17:55:09 CST 2021

janus 0.11.6
```

ustreamer installation steps
```
make WITH_OMX=1 WITH_JANUS=1
cp janus/libjanus_ustreamer.so /usr/lib/ustreamer/janus/
ln -s src/ustreamer.bin /usr/bin/ustreamer
./ustreamer -v
4.10
```

I don't know what I missed
我不知道我错过了什么
The HTTPS page can be logged in and opened normally, but there is no webrtc switching function
The browser supports webrtc

mdevaev, 2021-12-12:
`Dec 12 18:06:55 j3r-k kvmd-janus[500]: kvmd.apps.janus.runner            INFO --- => [ERR] [janus.c:main:5495] #011Couldn't load transport plugin 'libjanus_websockets.so': /usr/lib/janus/transports/libjanus_websockets.so: undefined symbol: lws_validity_confirmed`
Too old libwebsockets I guess

TelDragon, 2021-12-12:
Wow, your reply is so timely

libjanus_websockets.so  Too old？

janus version 0.11.6

TelDragon, 2021-12-12:
I'll check the system files
```
dpkg -l | grep janus
ii  janus                           0.6.1-1                              armhf        general purpose WebRTC gateway
ii  janus-dev                       0.6.1-1                              armhf        general purpose WebRTC gateway - development files
```

I'll try apt get remove and use 0.11 again 6 compilation

TelDragon, 2021-12-12:
Thank you. This problem has been solved

```
git clone https://libwebsockets.org/repo/libwebsockets
cd libwebsockets
# If you want the stable version of libwebsockets, uncomment the next line
# git checkout v3.2-stable
mkdir build
cd build
# See https://github.com/meetecho/janus-gateway/issues/732 re: LWS_MAX_SMP
# See https://github.com/meetecho/janus-gateway/issues/2476 re: LWS_WITHOUT_EXTENSIONS
cmake -DLWS_MAX_SMP=1 -DLWS_WITHOUT_EXTENSIONS=0 -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_C_FLAGS="-fpic" ..
make && sudo make install
```

```
/usr/bin/python3 /usr/local/bin/kvmd-janus --run
kvmd.apps.janus.runner            INFO --- Starting Janus Runner ...
kvmd.apps.janus.runner            INFO --- Got new _Netcfg(nat_type='Full Cone NAT', src_ip='192.168.1.177', ext_ip='119.248.23.120', stun_host='stun.l.google.com', stun_port=19302)
kvmd.apps.janus.runner            INFO --- Starting Janus ...
kvmd.apps.janus.runner            INFO --- Started Janus pid=813: ['/usr/bin/janus', '--disable-colors', '--plugins-folder=/usr/lib/ustreamer/janus', '--configs-folder=/etc/kvmd/janus', '--interface=192.168.1.177', '--stun-server=stun.l.google.com:19302']
kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
kvmd.apps.janus.runner            INFO --- => Compiled on:  Sun 12 Dec 19:20:41 CST 2021
kvmd.apps.janus.runner            INFO --- => Logger plugins folder: /usr/lib/janus/loggers
kvmd.apps.janus.runner            INFO --- => [WARN]    Couldn't access logger plugins folder...
kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
kvmd.apps.janus.runner            INFO --- => Starting Meetecho Janus (WebRTC Server) v0.11.6
kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
kvmd.apps.janus.runner            INFO --- => Checking command line arguments...
kvmd.apps.janus.runner            INFO --- => Debug/log level is 4
kvmd.apps.janus.runner            INFO --- => Debug/log timestamps are disabled
kvmd.apps.janus.runner            INFO --- => Debug/log colors are disabled
kvmd.apps.janus.runner            INFO --- => Using 192.168.1.177 as local IP...
kvmd.apps.janus.runner            INFO --- => Token based authentication disabled
kvmd.apps.janus.runner            INFO --- => Initializing recorder code
kvmd.apps.janus.runner            INFO --- => RTP port range: 20000 -- 40000
kvmd.apps.janus.runner            INFO --- => Initializing ICE stuff (Full mode, ICE-TCP candidates disabled, half-trickle, IPv6 support enabled)
kvmd.apps.janus.runner            INFO --- => ICE port range: 20000-40000
kvmd.apps.janus.runner            INFO --- => [WARN] mDNS resolution disabled, .local candidates will be ignored
kvmd.apps.janus.runner            INFO --- => STUN server to use: stun.l.google.com:19302
kvmd.apps.janus.runner            INFO --- => >> 172.217.213.127:19302 (IPv4)
kvmd.apps.janus.runner            INFO --- => Testing STUN server: message is of 20 bytes
kvmd.apps.janus.runner            INFO --- => >> Our public address is 119.248.23.120
kvmd.apps.janus.runner            INFO --- => Crypto: OpenSSL >= 1.1.0
kvmd.apps.janus.runner            INFO --- => [WARN] The libsrtp installation does not support AES-GCM profiles
kvmd.apps.janus.runner            INFO --- => No cert/key specified, autogenerating some...
kvmd.apps.janus.runner            INFO --- => Fingerprint of our certificate: BE:4C:62:0D:E1:2A:A7:D4:26:36:7F:1F:62:BA:4F:A7:E4:67:0B:F5:24:76:91:92:7B:7A:74:AB:3E:F3:0B:F5
kvmd.apps.janus.runner            INFO --- => [WARN] Data Channels support not compiled
kvmd.apps.janus.runner            INFO --- => Event handlers support disabled
kvmd.apps.janus.runner            INFO --- => Plugins folder: /usr/lib/ustreamer/janus
kvmd.apps.janus.runner            INFO --- => Joining Janus requests handler thread
kvmd.apps.janus.runner            INFO --- => Loading plugin 'libjanus_ustreamer.so'...
kvmd.apps.janus.runner            INFO --- => Sessions watchdog started
kvmd.apps.janus.runner            INFO --- => == ustreamer -- Initializing plugin ...
kvmd.apps.janus.runner            INFO --- => == ustreamer -- Reading config file '/etc/kvmd/janus/janus.plugin.ustreamer.jcfg' ...
kvmd.apps.janus.runner            INFO --- => [WARN] The 'janus.plugin.ustreamer' plugin doesn't implement any callback for RTP/RTCP/data... is this on purpose?
kvmd.apps.janus.runner            INFO --- => Transport plugins folder: /usr/lib/janus/transports
kvmd.apps.janus.runner            INFO --- => Loading transport plugin 'libjanus_websockets.so'...
kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected
kvmd.apps.janus.runner            INFO --- => [WARN] libwebsockets has been built without IPv6 support, will bind to IPv4 only
kvmd.apps.janus.runner            INFO --- => libwebsockets logging: 0
kvmd.apps.janus.runner            INFO --- => Websockets server started (UNIX socket /run/janus-ws.sock)...
kvmd.apps.janus.runner            INFO --- => JANUS WebSockets transport plugin initialized!
kvmd.apps.janus.runner            INFO --- => WebSockets thread started
kvmd.apps.janus.runner            INFO --- => [WARN] libnice version outdated: 0.1.14 installed, at least 0.1.16 recommended. Notice the installed version was checked at build time: if you updated libnice in the meanwhile, re-configure and recompile to get rid of this warning
```

mdevaev, 2021-12-12:
Welcome

closed 2021-12-12 by mdevaev

TelDragon, 2021-12-13:
@mdevaev 
Thank you for your support
But what puzzles me is that my web page still can't show webrtc switching

Nginx has no error log

nginx add juans.conf
kvmd.ctx-http.conf
```
upstream kvmd {
        server unix:/run/kvmd.sock fail_timeout=0s max_fails=0;
}

upstream ustreamer {
        server unix:/run/ustreamer.sock fail_timeout=0s max_fails=0;
}

upstream janus-ws {
        server unix:/run/janus-ws.sock fail_timeout=0s max_fails=0;
}
```

kvmd.ctx-server.conf
```
.............

location /janus/ws {
        rewrite ^/janus/ws$ / break;
        rewrite ^/janus/ws\?(.*)$ /?$1 break;
        proxy_pass http://janus-ws;
        include /etc/kvmd/nginx/loc-proxy.conf;
        include /etc/kvmd/nginx/loc-websocket.conf;
}

location = /share/js/kvm/janus.js {
        alias /usr/share/janus/javascript/janus.js;
        include /etc/kvmd/nginx/loc-nocache.conf;
}

location = /share/js/kvm/adapter.js {
        alias /usr/share/janus/javascript/adapter.js;
        include /etc/kvmd/nginx/loc-nocache.conf;
}
```

Check JS
```
ls -ll /usr/share/kvmd/web/share/js/kvm/janus.js 
-rw-r--r-- 1 root root 129758 Dec 12 20:16 /usr/share/kvmd/web/share/js/kvm/janus.js
 ls -ll /usr/share/kvmd/web/share/js/kvm/adapter.js 
-rw-r--r-- 1 root root 128269 Dec 12 20:16 /usr/share/kvmd/web/share/js/kvm/adapter.js

ls -ll /usr/share/janus/javascript/janus.js 
-rw-r--r-- 1 root root 129758 Dec 12 19:21 /usr/share/janus/javascript/janus.js
 ls -ll /usr/share/janus/javascript/adapter.js 
-rw-r--r-- 1 root root 128269 Jul 31 20:14 /usr/share/janus/javascript/adapter.js
```

Because there is no error log, I can't trace it
Is there any configuration I missed?

TelDragon, 2021-12-13:
The package of kvmd webtermis missing from the kvmd repository

```
manifest.yaml 
name: Terminal
description: Open terminal in a web browser
icon: extras/webterm/terminal.svg
path: extras/webterm/ttyd
daemon: kvmd-webterm
place: 10

nginx.ctx-http.conf 
upstream ttyd {
        server unix:/run/kvmd/ttyd.sock fail_timeout=0s max_fails=0;
}

nginx.ctx-server.conf 
location /extras/webterm/ttyd {
        rewrite ^([^.\?]*[^/(/ws)])$ $1/ permanent;
        rewrite ^/extras/webterm/ttyd$ / break;
        rewrite ^/extras/webterm/ttyd\?(.*)$ ?$1 break;
        rewrite ^/extras/webterm/ttyd/(.*)$ /$1 break;
        proxy_pass http://ttyd;
        include /etc/kvmd/nginx/loc-proxy.conf;
        include /etc/kvmd/nginx/loc-websocket.conf;
        include /etc/kvmd/nginx/loc-login.conf;
        include /etc/kvmd/nginx/loc-nocache.conf;
}
```