pikvm/pikvm #569: Incorporate support for Rock 3A SBC

SleepIT, 2021-12-13:
The Rock 3A SBC (which, confusingly, is newer than the Rock Pi 4A/B) is GPIO and CSI compatible with the RPI 4.  Unlike the RPI 4, however, it is actually available for purchase. 

**Is your feature request related to a problem? Please describe.**
The only problem is the lack of Raspberry Pi 4 availability.

**Describe the solution you'd like**
For PiKVM to incorporate support for the Rock 3A SBC.

**Describe alternatives you've considered**
I've looked at many alternative SBCs and the Rock 3A seems to be the best candidate to work with PiKVM w/ HDMI-CSI2 adapter.  Most other SBCs have an incompatible CSI interface and/or alternative GPIO layout.  Additionally, it has a selectable (host/device) OTG USB port which eliminates the need for custom cabling.  It is also a current SBC (released within the last 6 months) so should have availability for some time.

**Additional context**
I have some on order and would like to assist with testing/debugging/etc.

mdevaev, 2021-12-13:
Right now we don't have the resources to support alternative boards. There were several attempts at porting, but they all ended either with nothing or individually working hardware. Considering that even with one Raspberry, there are periodically problems with the kernel, supporting alternative platforms out of the box will create twice as many problems for us. So my position on this is: we don't do it ourselves, we won't provide support, but we won't stop you and are ready to accept patches, provided that they don't break anything in the existing ecosystem for Raspberry Pi. Since I am the only software developer, I definitely don't have time for alternative platforms, sorry.

SleepIT, 2021-12-14:
Understood.  Will see if I can get things working!

closed 2021-12-14 by SleepIT

KarimGeiger, 2021-12-15:
@SleepIT I would be interested in that as well. If you manage to get any progress, please keep us posted. I’m sure there are others who would benefit from broader compatibility as well. 

Unfortunately, I myself aren’t near skilled enough to help with this and I totally understand mdevaevs position. I can only imagine how difficult it would be to support multiple platforms as a single developer.

SleepIT, 2021-12-16:
@KarimGeiger I will update here.  Currently waiting for the hardware to arrive, then see how it goes from there.  It will definitely be a learning process for me, but here's hoping!