pikvm/pikvm #570: Migrate project browser compatibility check webrtc

TelDragon, 2021-12-15:
Dear developer, I have successfully run pikvm, ustreamer, Janus and ttyd into the raspberry pie operating system
尊敬的开发者您好，目前我已经将PIKVM、ustreamer、janus、ttyd成功的运行再了树莓派操作系统中

But I encountered a problem about browser compatibility (doubt, because there is no error message)
但是我遇到一个关于浏览器兼容性问题（怀疑，因为没有报错信息）

I can use Firefox browser to open MJPEG / HTTP and h.264/webrtc images, and operate perfectly
我使用firefox 浏览器均可以打开以MJPEG/HTTP 和 H.264/WebRTC 画面，并且完美操作

version
```
KVMD: 3.50
Streamer: 4.10 (ustreamer)

    HAS_PDEATHSIG: Yes
    WITH_GPIO: No
    WITH_OMX: Yes
    WITH_PTHREAD_NP: Yes
    WITH_SETPROCTITLE: Yes

Linux kernel:

    Machine: armv7l
    Release: 5.10.17-v7l
    Version: #1 SMP Thu Oct 14 11:07:58 HKT 2021
```
However, using Google Chrome, Microsoft edge and browser cannot connect to h.264/webrtc And you can't see any error logs
但是使用google chrome、microsoft edge、浏览器均无法正常连接H.264/WebRTC. 并且也看不到任何的错误日志

At first, I suspected the Janus version, 0.11 6 and 0.11 All of them have been tried, and the problem still exists.
我开始是怀疑janus版本问题，0.11.6 和 0.11.5 全部都试过，问题仍然存在。

I will your OS image v2-hdmi-rpi4-latest img. The two JS "adapter. JS" and "Janus. JS" in XZ have also been overwritten, but there is still no change
我将您的os镜像 v2-hdmi-rpi4-latest.img.xz 内的两个js “adapter.js”、“janus.js” 也进行过覆盖，仍然无任何改变

This should not be a problem of your project, but I hope you can give me some inspection suggestions to help me solve this problem.
这应该不是您项目的问题，但是很期望您能给些检查建议，来帮助我解决这个问题。

This is a log that is accessed by Firefox browser, and can be connected normally after 10+ seconds, because I have access to Google stun in Chinese mainland. There may be some restrictions.
这是使用firefox 浏览器访问的日志，等待10+秒后可以正常连接，因为我是中国大陆用户 访问google stun可能有某些限制
```
Dec 15 18:47:11 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [3908091422008967] WebRTC resources freed; 0xb1b016c0 0xb1b01258
Dec 15 18:47:11 pi kvmd[572]: kvmd.apps.kvmd.server             INFO --- Removed client socket: WsClient(id=3022403728, stream=True); clients now: 0
Dec 15 18:47:11 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /ws HTTP/1.0' => 101; size=0 --- referer='-'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:11 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Removing session 0xb1b017e8 ...
Dec 15 18:47:11 pi kvmd[572]: kvmd.apps.kvmd.streamer           INFO --- Planning to stop streamer in 10.00 seconds ...
Dec 15 18:47:11 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink closed
Dec 15 18:47:11 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='-'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:11 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='-'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:11 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /info?fields=meta,extras HTTP/1.0' => 200; size=1371 --- referer='https://192.168.1.176/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:11 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:12 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / 192.168.1.51] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/kvm/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: kvmd.apps.kvmd.server             INFO --- Registered new client socket: WsClient(id=3022464688, stream=True); clients now: 1
Dec 15 18:47:17 pi kvmd[572]: kvmd.apps.kvmd.streamer           INFO --- Streamer stop cancelled
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/share/js/kvm/stream.js'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.176/share/js/kvm/janus.js'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd[572]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='-'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:95.0) Gecko/20100101 Firefox/95.0'
Dec 15 18:47:17 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => Creating new session: 2955365698715182; 0xb0600768
Dec 15 18:47:17 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => Creating new handle in session 2955365698715182: 2147066251988825; 0xb0600768 0xb1b04c40
Dec 15 18:47:17 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Creating session 0xb1b014d0 ...
Dec 15 18:47:17 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [2147066251988825] Creating ICE agent (ICE Full mode, controlling)
Dec 15 18:47:17 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink opened; reading frames ...
Dec 15 18:47:19 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [2147066251988825] Waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
Dec 15 18:47:20 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [2147066251988825] Still waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
Dec 15 18:47:21 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [2147066251988825] Still waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
Dec 15 18:47:22 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [2147066251988825] Still waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
[... 42 lines cut ...]
Dec 15 18:54:43 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [1629767705204615] Waited 5s for candidates, that's way too much... going on with what we have (WebRTC setup might fail)

systemctl status kvmd-janus
● kvmd-janus.service - PiKVM - Janus WebRTC Gateway
   Loaded: loaded (/lib/systemd/system/kvmd-janus.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-12-15 18:46:47 CST; 10min ago
 Main PID: 1733 (/usr/bin/python)
    Tasks: 29 (limit: 2827)
   CGroup: /system.slice/kvmd-janus.service
[... cut ...]
Dec 15 18:57:36 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [1673188868844033] Still waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for 
Dec 15 18:57:37 pi kvmd-janus[1733]: kvmd.apps.janus.runner            INFO --- => [WARN] [1673188868844033] Waited 5s for candidates, that's way too much... going on with what we have (WebRTC setup migh
```
If you replace kvmd Janus with kvmd Janus static, the effect is the same, but it doesn't pass Google stun. It just increased my local connection speed
如果将kvmd-janus 替换为kvmd-janus-static 效果一样，只不过它不通过google stun 了。

arch1mede, 2021-12-16:
IMO: This is not an issue as this project does not support any other OS than arch arm linux. Unofficial testing using rasbian buster, all browsers worked.

mdevaev, 2021-12-16:
For what purpose do you need all this at all? What exactly are you doing?

TelDragon, 2021-12-17:
Dear author, according to your idea, we have made a demo based on CM4, integrating tc358743

But most of our software is completed under the raspberry pie system or Debian system. We are not very familiar with arch arm linux...

mdevaev, 2021-12-17:
We do not provide support outside of our products.

closed 2021-12-17 by mdevaev

TelDragon, 2021-12-20:
Anyway, I would like to thank the respected author

This problem has been solved because Janus relies on the low libnice version,
libnice (at least v0.1.16 suggested, v0.1.18 recommended)