pikvm/pikvm #572: support usb cameras

linuxsky, 2021-12-16:
At present, KVM web supports displaying HDMI input, but I often want to have a camera to see the surrounding environment. If I plug a USB camera into KVM, can I display the camera content on the web and switch freely with the desktop content. If yes, how should I set it?

arch1mede, 2021-12-16:
Please use the USB image and follow the V2 USB setup using the lower left usb2.0 port per the instructions.

linuxsky, 2021-12-17:
> Please use the USB image and follow the V2 USB setup using the lower left usb2.0 port per the instructions.

How can I set it so that the USB camera can also be displayed on the web, similar to the desktop.

arch1mede, 2021-12-17:
Hi, I don't feel this is the correct forum for further troubleshooting, please join the Discord and you can get further assistance.

In a nut shell you are replacing the usb dongle with the usb webcam but you will not be able to view them the same as you can on the desktop.

arch1mede, 2021-12-17:
While this is possible, it is not usable, others have tried it....this is also not something that we will support as it requires some advanced linux skills.

mdevaev, 2021-12-23:
What you want is a special case of multimonitor support. We have a plan to do this in the future, but it's not a priority right now. I can suggest you to run another ustreamer with some scripts or through systemd, and then throw it through nginx.

closed 2021-12-23 by mdevaev