pikvm/pikvm #573: What would be needed to have pikvm with ecozoo 4 port switch to work with mesh central?

weismannweb, 2021-12-16:
What would be needed to have pikvm with ecozoo 4 port switch to work with mesh central?

https://www.youtube.com/watch?v=jWgyH9I3xcI

This probably belongs in the mesh repo feature requests but I am putting it here to see what might be needed on the pikvm side?

arch1mede, 2021-12-16:
My feedback: Not likely. MeshCentral is a Raritan proprietary enterprise solution. PiKVM is a 1:1 solution

Technically, ezcoo is not part of the PiKVM ecosystem, its just got a ctl port that arch linux can access. There would be more like it if they all had ctl ports but most do not.

weismannweb, 2021-12-16:
Thanks for the answer!  Oh well.  Was really just a nicety to have one portal.

closed 2021-12-16 by mdevaev