pikvm/pikvm #576: Cannot connect wifi in country=KR

luxflow, 2021-12-18:
**Describe the bug**
- I cannot connect wifi 2.4ghz in country=KR
- I cannot list wifi 5.0ghz
- I tested official [rpios](https://www.raspberrypi.com/software/operating-systems/)(Release date: October 30th 2021), both 2.4ghz, 5ghz works well on this image on same machine. so I think it is not a hardware problem
- There was a country=KR specific [bug](https://forums.raspberrypi.com/viewtopic.php?t=314361&sid=2ad3b1c3723ce71fda167c46caeaf266&start=25) in brcmfmac firmware which seems to already patched latest rpios image, replace newer firmware doesn't solve problem

**To Reproduce**
Steps to reproduce the behavior, like:
1. follow https://docs.pikvm.org/wifi/ guide
2. add "country=KR" first line in wpa_supplicant-wlan0.conf

**Expected behavior**
both 2.4ghz, 5ghz wifi works well

**Screenshots**
not needeed

**Desktop (please complete the following information):**
not needeed

**PiKVM info:**
  - Raspberry Pi board version RPI4 + pikvm v3 hat (kickstarter)

**Additional context**
I suspect brcmfmac driver problem.
rpios brcmfmac driver version is `5.10.63-v71+`

luxflow, 2021-12-18:
[D.zip](https://github.com/pikvm/pikvm/files/7738876/D.zip)
what weird thing is sometimes wifi connect successfully (see log)

arch1mede, 2021-12-24:
My 2 cents:

Dev's are working on getting a new version released that MAY resolve this issue, in the mean time I'm afraid you will need to wait.

With that said, this is not a PiKVM issue, this is an underlying OS (Arch Arm) issue, so should be directed towards rpi foundation forums or arch arm discord/forums. You cannot be the only one that has a rpi and having connection issues. I would also reach out to a local maker shop and ask them about your issues and ask how they were able to solve this.

luxflow, 2021-12-28:
I suspect arch arm issue not rpios issue since KR region and hdmi interference issue is resolved on rpios
(BTW is there anyone who having hdmi, wifi interference issue on pikvm?)
I somewhat agree this is OS problem not pikvm issue but I created this issue for reporting and sharing info who having similar wifi problem since I cannot find opened wifi issue

it is ok to close

closed 2022-01-15 by mdevaev