pikvm/pikvm #577: Limited Available Resolutions

somejavadev, 2021-12-20:
**Describe the bug**
Currently I have a Hat v3 with a raspberry Pi4 8Gb but for some reason I cannot increase the display resolution to 1080p, I am restricted to a max resolution of 1027x768 within the host OS.

**To Reproduce**
  
1. Setup the PiKVM according to the Basic Setup and First Steps
2. Configure the PiKVM with multiport setup.
3. View available resolutions in Windows OS

**Expected behavior**
I would like to increase the resolution within the host OS to the supported 1080p resolution.

**Screenshots**

[image]   
   
[image]

**Desktop (please complete the following information):**
 - OS: Windows Server 2019
 - Browser: chrome

**PiKVM info:**
  - Raspberry Pi board version:   
RPi 4 8Gb   
  - PiKVM platform     
Hat v3 (kickstarter)  
  - Video capture type [e.g. CSI bridge]  
Hat v3 (kickstarter)    
  - KVMD version: `pacman -Q | grep kvmd`    
kvmd 3.50-1  
kvmd-fan 0.1-1  
kvmd-oled 0.8-1  
kvmd-platform-v3-hdmi-rpi4 3.50-1  
kvmd-webterm 0.41-1  

  - uStreamer version: `pacman -Q | grep ustreamer`  
ustreamer 4.10-1  
  - Linux kernel: `uname -a`  
Linux pikvm 5.10.52-14-ARCH #1 SMP Thu Dec 2 09:40:42 MSK 2021 armv7l GNU/Linux

**Additional context**
I am running the PiKVM in [multiport mode](https://docs.pikvm.org/multiport/) with the ezCoo SW41HA HDMI 4x1 switch.   
The Windows server motherboard is the P8Z68-V with the [most up to date Bios ](https://www.asus.com/SupportOnly/P8Z68V/HelpDesk_BIOS/), this is quite an old motherboard, initially I couldn't enter the BIOS via the PiKVM since the keyboard wasn't active, I rectified it by updating to the latest version.    
   
Having a read through the FAQ I don't see anywhere on how I could set the resolution to 1080p on the PiKVM. The system menu in the browser don't have such an option.   
   
Seeing the Keyboard/Video and mouse is picked up in the BIOS can this still be an EDID issue?

mdevaev, 2021-12-20:
Did you try without ezcoo switch?

somejavadev, 2021-12-20:
@mdevaev Awesome to see you on the project, thanks for this awesome product. I tried it without the ezcoo, connecting the HDMI from the PC directly into the PiKVM and it still seems to only have the limited options. I restarted the PC after swapping out the ezcoo to double check without any luck.

mdevaev, 2021-12-20:
Have you tried default EDID?
```
rw
cp /usr/share/kvmd/configs.default/kvmd/tc358743-edid.hex /etc/kvmd/
reboot
```
Next reboot the server.

somejavadev, 2021-12-21:
@mdevaev, Thanks I gave that a try but it still seems stuck at 1027x768, what I did noticed during the boot the screen resolution jumps between 640x480 and 800x600 and then later on to 1027x768, although I have seen this with other KVM's as well.  
  
The monitor properties are as follow:  
  
[image]  
  
[image]  
  
Interesting what I noticed if I apply the `1280x1024 as preferred. Useful for Gigabyte GA-H77-DS3H` EDID it works for using the 1280x1024 resolution, When I apply the `1920x1080 as preferred. Useful for Gigabyte GA-H77-DS3H or Intel NUC` using the temporary method the PC is stuck at boot but while it's stuck if I apply the default EDID it gets unstuck and the frame size in the UI is 1080p but you still can't choose 1080p from the display settings.   
   
[image]   
   
I downloaded and installed the AW EDID Editor but I have no idea on how to go about using it.  
I also tried plugging out and back in the HDMI cables without any luck.

somejavadev, 2021-12-21:
I have found this: https://edid.tv/  will see if I can try some other EDIDs

mdevaev, 2021-12-21:
I've never seen such things and it doesn't look like it's a problem on the part of PiKVM. Maybe the video card driver in Windows is weird. Do you have any other computer on which you could check this?

EDID is just data that indicates which resolutions the monitor supports (in our case, PiKVM). Sometimes hosts ignore this and try to work in a resolution that is not supported on our part, but our EDID clearly says that it can work with 1080p50 and others.

Based on the resolutions that Windows offers you, I see that they are all aspect ratio 4:3. I don't know Windows too well, but it is possible that somewhere there is a setting that forces the use of 4:3, because 1080p and 720p are 16:9. I believe that all 16:9 on this host can simply disabled or so.

somejavadev, 2021-12-21:
Thanks for the response, let me have a look. I am gonna leave the ticket open for now and will post back when I manage to resolve the issue. It might be that it's an issue with the video card drivers on windows like you said.

somejavadev, 2021-12-21:
Finally success. I verified that it was a driver issue by booting up an Ubuntu Live CD:  
   
[image]  
  
After determining that, I couldn't find the Intel Driver for server 2019, after doing some searches I found this page on [reddit](https://www.reddit.com/r/sysadmin/comments/drf58q/installing_intel_gpu_drivers_on_a_windows_2019/), I then download the 64Bit graphics drivers for windows 8.1 from the manufactures website and used this command to install it:  
```ps
$Driverpath = "C:\Drivers\Intel_Graphics"

$driversCollection = Get-ChildItem -Path $Driverpath -Filter "*.inf" -Recurse -ErrorAction SilentlyContinue | Select-Object -ExpandProperty Fullname 

foreach($driver In $driversCollection){
	#Add and install driver package
	pnputil.exe -i -a $driver 
	}
```

closed 2021-12-21 by somejavadev

mdevaev, 2021-12-21:
Awesome!

@arch1mede could you document this in the FAQ?

arch1mede, 2021-12-30:
This was added to the FAQ under video problems.