pikvm/pikvm #578: documentation update - "Flashing the OS image"

vbooka1, 2021-12-21:
I suggest updating this document: https://docs.pikvm.org/flashing_os/

This part:

> Using Linux CLI
> Decompress and flash the image. Be careful when choosing your device path:
> # dd if=v2-hdmi-rpi4-latest.img of=/dev/mmcblkX

Add:

> # # or, if using USB SD card reader:
> # dd if=v2-hdmi-rpi4-latest.img of=/dev/sdX 
> # # where X is the device letter, not the partition number -- use /dev/sdX, not /dev/sdX1

It took me a while to understand why rPI does not boot after I flashed the image to /dev/sdb1, twice 🤭

arch1mede, 2021-12-24:
Thank you for the feedback - We will discuss if this would be worth adding in. The logic is that if you choose to do the dd route that you have the required linux skills to know about the lettering differences, hence the "Be careful when choosing your device path" verbiage.

mdevaev, 2021-12-27:
Changed: `Be careful when choosing the device path, it may be different on your OS`

closed 2021-12-27 by mdevaev

vbooka1, 2021-12-28:
> Changed: Be careful when choosing the device path, it may be different on your OS

Still not clear I meant to add a comment about using the device name itself, without the partition number - e.g. "Use the device name itself, not the partition number -- like `/dev/sdx`, not `/dev/sdx1` "

commit fbfa9350 (2021-12-27): Fixed #578