pikvm/pikvm #580: CD Read Fail when custom MECM task sequence media connected

aggietallboy, 2021-12-21:
**Describe the bug**
Booting to the Bootable task sequence ISO, system boots into main Microsoft Endpoint Configuration Manager screen, but returns "unable to read task sequence configuration disk. For more information contact your system administrator or helpdesk operator".

Same ISO used with Rufus, produces GPT bootable disk which works fine to run task sequences.

**To Reproduce**
1. Create bootable task sequence media in SCCM/MECM
2. Copy iso to pikvm (select image to upload...)
3. Connect drive to server
4. boot
5. fail

**Expected behavior**
Welcome to the Task Sequence Wizard

**Screenshots**
Failure:
[image]

Success/Normal:
[image]

**Desktop (please complete the following information):**
Chrome browser running in app mode

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] - Raspberry Pi 4 Model B Rev 1.4 - 8gb
  - PiKVM platform [e.g. v2-hdmi]  - v3-hdmi-rpi4-latest.img
  - Video capture type [e.g. CSI bridge] - PiKVM Hat from Kickstarter campaign
  - KVMD version: `pacman -Q | grep kvmd` - 3.48
  - uStreamer version: `pacman -Q | grep ustreamer` 4.9
  - Linux kernel: `uname -a` 5.10.52-13-ARCH

**Additional context**
Add any other context about the problem here.

arch1mede, 2021-12-24:
While I do not use Windows PE, I believe it has the same limitations as the other windows installs: Please see [here](https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive)

arch1mede, 2021-12-29:
Ping, does this work for you?

mdevaev, 2022-04-30:
No answer from user.

closed 2022-04-30 by mdevaev