pikvm/pikvm #583: Set different interface metric for eth0 and wlan0

nvtkaszpir, 2021-12-26:
**Is your feature request related to a problem? Please describe.**
Right now both interfaces get the same routing metric:
```text
[root@pikvm /]# ip r
default via 192.168.1.1 dev wlan0 proto dhcp src 192.168.1.37 metric 1024 
default via 192.168.1.1 dev eth0 proto dhcp src 192.168.1.38 metric 1024 
```
But it would be better to favour cable interface for dynamic routing.

**Describe the solution you'd like**
This can be solved by editing /etc/systemd/network/eth0.network and adding RouteMetric into DHCP section like below (last line):
```text
[Match]
Name=eth0

[Network]
DHCP=yes
DNSSEC=no

# Use same IP by forcing to use MAC address for clientID
[DHCP]
ClientIdentifier=mac
RouteMetric=10

```
and running `systemctl restart systemd-networkd` will automatically reconfigure interfaces:
```text
[root@pikvm /]# systemctl restart systemd-networkd
[root@pikvm /]# ip r
default via 192.168.1.1 dev eth0 proto dhcp src 192.168.1.38 metric 10 
default via 192.168.1.1 dev wlan0 proto dhcp src 192.168.1.37 metric 1024 
```

For statically assigned addresses it will look a bit different, though, but that's another story. More on that in systemd.network.5 (link below)

**Describe alternatives you've considered**
None.

**Additional context**
https://man.archlinux.org/man/systemd.network.5

mdevaev, 2021-12-27:
Perhaps we will change OLED script and Janus runner to get right default network interface. Hint for me: netifaces + /proc/net/route

nvtkaszpir, 2021-12-27:
Right now if two interfaces are set as default gateway and they have the same metric then the first one registered in the system is returned.
If metric value is changed on the fly (as described in first comment of this github issue) then interface with lowest metric is returned, no need to restart kvmd-oled service.

mdevaev, 2021-12-27:
Done. It will be available in the next OS release.

closed 2021-12-27 by mdevaev