pikvm/pikvm #585: flip input screen

DBursch78, 2021-12-28:
**Is your feature request related to a problem? Please describe.**
I am using a pikvm hat v3 and for some reasons my hdmi input is rotate to upside down

**Describe the solution you'd like**
I guess i need option like "rotate input video" or the output 

**Describe alternatives you've considered**
Any option to flip the output screen so the mouse works right but the screen as to flip

arch1mede, 2021-12-29:
Please join Discord in #support channel, id like to discuss as I have never heard of this before.

mdevaev, 2022-01-04:
@DBursch78 is it solved in the Discord or required my help?

mdevaev, 2022-01-10:
Sup?

DBursch78, 2022-01-10:
I go into it next week, I’m on vacation 

> 
> 
> Sup?

arch1mede, 2022-01-26:
@DBursch78 boop?

arch1mede, 2022-02-01:
@DBursch78 boop, boop?

mdevaev, 2022-03-16:
Apparently, this is no longer needed.

DBursch78, 2022-03-16:
I post a vid with that issue?!

> 
> 
> Closed #585.

mdevaev, 2022-03-16:
What kind of signal source are you using?

DBursch78, 2022-03-16:
Hdmi, it's  in the Settings from System, all Display’s are set to upside down. That’s the reason why I needed to rotate the stream.

> 
> 
> What kind of signal source are you using?

mdevaev, 2022-05-24:
This is not a best solution, but anyway. Edit file `/etc/kvmd/web.css` and add:
```css
video#stream-video {
    transform: rotate(180deg);
}
img#stream-image {
    transform: rotate(180deg);
}
```

closed 2022-05-24 by mdevaev

DBursch78, 2022-06-09:
Works perfect, good work, thank you!

mdevaev, 2022-06-09:
:ok_hand:

BhupiMAD, 2023-08-20:
While this solution works, the mouse on the screen is still not inverted as per the screen resolution. You see mouse going in opposite direction. Any fix for this?

mdevaev, 2023-08-20:
You can invert the mouse range. Run `kvmd -m` and find options tree with `mouse_x_range` and `mouse_y_range`.