pikvm/pikvm #589: Add support for power on/off button on gpio 3

Kal42, 2021-12-29:
**Describe the solution you'd like**
An option in /etc/kvmd/override.yaml, htat enable support of power on/off push button on gpio 3.
If enable, "kvmd" should handle the listen on gpio 3 and start the process of shutdown, if button is press more than X second (I think this should be parametrable)

**Describe alternatives you've considered**
I try to use the allready existing gpio configuration, but I am not able to make it works and I am not sur if it's possible.

mdevaev, 2021-12-29:
Already implemented. Use this override:
```yaml
kvmd:
    atx:
        power_switch_pin: 3
```

Kal42, 2021-12-29:
Sorry, I was not clear. It's for shutdown the rpi, not remote pc.

mdevaev, 2021-12-29:
Why you need it?

Kal42, 2021-12-30:
I have pc/server, that are not very acessible to put screen and keyboard, so pikvm is a very great project to get acccess to them, when there is issue. It will be the same for the pikvm, it will not be easy to have a screen/keyboard near it to properly shutdown it if needed. So it can be good, if I can shut it down properly, with a push button.

I have  done a small pyton script based on the work from Frederik Lindenaar :
```
#! /usr/bin/env python

#
# gpio_trigger.py - execute a command when a GPIO input pin changes
#
# Version 1.0, latest version, documentation and bugtracker available at:
#               https://gitlab.lindenaar.net/scripts/raspberrypi
#
# Copyright (c) 2019 Frederik Lindenaar
#
# This script is free software: you can redistribute and/or modify it under the
# terms of version 3 of the GNU General Public License as published by the Free
# Software Foundation, or (at your option) any later version of the license.
#
# This script is distributed in the hope that it will be useful but WITHOUT ANY
# WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR
# A PARTICULAR PURPOSE.  See the GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License along with
# this program.  If not, visit <http://www.gnu.org/licenses/> to download it.
#

from os import system
from RPi import GPIO

GPIO.setmode(GPIO.BCM)                                 # set GPIO number mode
GPIO.setup(3, GPIO.IN, pull_up_down=GPIO.PUD_UP)   # setup pin as input

ret = 0
while ret == 0:
    # wait for GPIO pin to be changed in the right direction
    if GPIO.wait_for_edge(3, GPIO.RISING, bouncetime=200,
                                                        timeout=-1):
        state = GPIO.input(3)
        # pin changed, if required check if it is stays the same long enough
        if (GPIO.wait_for_edge(3, GPIO.BOTH, bouncetime=200,
                                    timeout=5000 - 200) is None):
            ret = system('/usr/bin/poweroff')
            break
    else:                                               # exit if timeout
        ret = -1
        break

GPIO.cleanup(3)                                  # cleanup GPIO

exit(ret)                                               # and return the result
```

It's work if I launch it manually aftyer the pi start. But I am not able to make it works with autostart with systemd.

Here is my systemd also base on the work from Frederik Lindenaar :
```
#
# rpi_poweroff_button.service – Systemd service to implement a poweroff button
#
# Version 1.0, latest version, documentation and bugtracker available at:
#               https://gitlab.lindenaar.net/scripts/raspberrypi
#
# Copyright (c) 2019 Frederik Lindenaar
#
# This script is free software: you can redistribute and/or modify it under the
# terms of version 3 of the GNU General Public License as published by the Free
# Software Foundation, or (at your option) any later version of the license.
#
# This script is distributed in the hope that it will be useful but WITHOUT ANY
# WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR
# A PARTICULAR PURPOSE.  See the GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License along with
# this program.  If not, visit <http://www.gnu.org/licenses/> to download it.
#

# Please note this requires gpio_trigger.py installed in /usr/local/sbin, see:
# https://frederik.lindenaar.nl/2019/10/23/raspberry-pi-power-off-button.html

# To install copy to /etc/systemd/system/rpi_poweroff_button.service
# enable it to start during boot with:  systemctl enable rpi_poweroff_button
# temporarily enable with:              service rpi_poweroff_button start
# temporarily disable with:             service rpi_poweroff_button stop
# disable starting during boot with:    systemctl disable rpi_poweroff_button

[Unit]
Requires=kvmd.service
After=kvmd.service
Description=Power-off button on GPIO

[Service]
Type=idle
ExecStart=/usr/local/sbin/gpio_poweroff.py
Restart=always
RestartSec=30

[Install]
WantedBy=default.target
```

I have put to start it after kvmd, but it still start to early, as kvmd seems to take control over the gpio after.

And why gpio 3, it's because this gpio allow to put a push button to start the device without removing/putting back the power when we shutdown the pi (function WAKE_ON_GPIO in the bootloader)

Best regards,

mdevaev, 2021-12-30:
PiKVM uses a read-only file system, so it can simply be powered off at any time without any consequences. The only exception is the moment when you upload the image to the mass-storage drive emulator - you need to either cancel or wait for the upload to complete, after which you can pull the plug out of the socket.

Kal42, 2021-12-30:
For information, I have already broken sd card, with ro filesystem by shuting down the power directly.

arch1mede, 2021-12-30:
> For information, I have already broken sd card, with ro filesystem by shuting down the power directly.

Just adding in my 2 cents on this. This is not uncommon for SD card to do this especially cheapo SD cards.

mdevaev, 2021-12-30:
Poweroff would not have saved your card (it does not turn off the power of the memory card), the breakdown is not related to read-only. We use a lot of devices where it all works great. As said, you're just out of luck with a bad memory card. Buy another one class 10 from Samsung, for example, and everything will be fine.

closed 2022-01-04 by mdevaev