pikvm/pikvm #591: Mouse click on Android 11 / Samsung Tab 7 not working

wb7, 2022-01-01:
**Describe the bug**
When accessing PiKVM with the named tablet using Chrome oder Firefox, the mouse can be moved using the touchscreen or the attached touchpad, but clicking is not possible. Neither by tapping on the area, nor by clicking using the touchpad. The same happens on iOS 15 with iPad and iOS; clicking via touch-tap is not possible.

If an external bluetooth mouse is used, everything works as aspected.

arch1mede, 2022-01-04:
Based on my testing: The displayed window, which is larger than the tablet resolution, hides the mouse buttons and coding the mobile version is a low priority per a past ticket, you can gain this functionality if you enable vnc and use a vnc client. On my iPad Pro, i can see these mouse buttons.

bora-fsd, 2022-09-15:
I am experiencing this issue with PiKVM 3 as well (in my case, Android and Chrome)
Bluetooth mouse or tapping does not work in Chrome, however mouse moves are reflected in the host machine.
If I switch to Firefox, mouse clicks and tapping works.
I'm not sure what @arch1mede means by "mouse buttons" - I do not see any mouse buttons on the UI.

Would be great if this is prioritized!

ashao2devops, 2022-09-18:
Similar issue here, PiKVM Web UI on iPhone/IOS mouse click not working, I am able see the mouse point, mouse move, but can't click or touch/tap click. Can someone make it work please?

mdevaev, 2022-09-18:
I'm working on it. Just got a new iPad to debug.

mdevaev, 2022-10-06:
Fixed for iOS now.

mdevaev, 2022-10-06:
For Android will be available in the next release.

closed 2022-10-06 by mdevaev

si458, 2023-06-15:
may ï ask when this feature was fixed/released?
im using the latest pi zero w image and its not working but im guessing the image is too old and i cant update the pikvm?

mdevaev, 2023-06-16:
We don't support Zero W anymore because it was deprecated by the Arch ARM upstream. I recommend you to upgrade to Zero 2 W. It's inexpensive and works much better.

si458, 2023-06-16:
> We don't support Zero W anymore because it was deprecated by the Arch ARM upstream. I recommend you to upgrade to Zero 2 W. It's inexpensive and works much better.

Hi, I understand that, it's all I'm hearing everywhere, but you physically can't get ahold of a zero 2 w anywhere so I'm stuck with a load of zero W's, so is the a way to update pikvm manually or build it for the Zero W to include this patch to fix the touch on mobile? Sorry to sound snotty

mdevaev, 2023-06-18:
@si458 there is no way to do this. I can't say what specific changes were made. You can try to take the [web](https://github.com/pikvm/kvmd/tree/master/web) folder from git and put it in /usr/share/kvmd, but I don't promise that it will work. I have not maintained backward compatibility with the web ui.

si458, 2023-06-18:
@mdevaev just had a look, the fix was applied between version 3.153 and 3.148,
if i download the repo from 3.153 then copy over the web folder, and refrsh the webpage, 
it does indeed change the logo (so i no its updated),
but it never loads the page after login
the javascript console shoes an error 400 on a web request
`https://192.168.146.50/api/info?fields=auth,meta,extras`
`Object { error: "ValidatorError", error_msg: "Failed sub-validator on one of the item of ['auth', 'meta', 'extras']" }`
im assuming this is because the kvmd is missing a new feature that was introduced in 2022 ?