pikvm/pikvm #594: Paste: please add keymap for us-intl

j0xaf, 2022-01-03:
**Is your feature request related to a problem? Please describe.**

I am using `setxkbmap us altgr-intl` as keymap on my target. When pasting text containing Umlauts (äöüß etc) using the paste popup, the umlauts are omitted.

**Describe the solution you'd like**

 Could you please add `us altgr-intl` as option in the paste popup? Or is there a way for myself to add it?

Btw, PiKVM with the Hatv3 is the best KVM solution I have used so far! Thank you for creating!

mdevaev, 2022-01-03:
Sup. Where I can get this keymap?

j0xaf, 2022-01-04:
Thank you! I couldn't find one online, so I generated it using `qemu-keymap` myself.

See https://github.com/j0xaf/keymap-us-altgr-intl/blob/main/keymap-us-altgr-intl for the generated file.

mdevaev, 2022-01-04:
Added, will be available in the next release.

closed 2022-01-04 by mdevaev