pikvm/pikvm #596: How to reduce latency on background Web UI

wcck, 2022-01-05:
Dear Authors 

Situation : 
I designed a script for type some text in notepad on Web UI but when I minimize the Web UI the typed text will appear heavy latency.
e.g. 
**Normal case on script** : 1 2 3 4 5 6
**Latency case on script :** 11111 22222 33333 44444 55555 66666

In this case can i fix it or reduce the latency?
Thanks

mdevaev, 2022-01-05:
Hello. I believe that the problem is related to a decrease in the speed of script execution by the browser, that is, this is not a PiKVM problem, but a feature of the browser that we cannot influence.

However, I think you can work around this problem if you use the API directly: https://docs.pikvm.org/api/#websocket-events

wcck, 2022-01-06:
Hi @mdevaev 

I want to double confirm with you.
If I using **import websocket** and **develop python script** whether can solve this issue(https://github.com/pikvm/pikvm/issues/598)
Thanks

closed 2022-01-06 by mdevaev

wcck, 2022-01-06:
Hi @mdevaev 
I solved it.
Solution : Using websocket and develop python script to control Web UI.
It didn't appear latency again when minimize the bower.
Thanks

mdevaev, 2022-01-06:
Welcome!