pikvm/pikvm #599: wubGUI target window oversized by just a few pixel

VoeGalore, 2022-01-08:
I have PiKVM v3 Kickstarter version and upgraded to the latest (as of 2022-01-08), but this issue is present in my previous v2 build from early last year.

The webGUI target video feed is automatically resized to target resolution, you may also use the "." button in the top right corner of the video feed window to manually resize it back to target resolution (if target resolution is smaller than the screen you are viewing it with).
However the video feed has always been non-pixel-perfect. Noticeably there are blurry areas indicating the displaying resolution not completely matching the target(video feed) resolution. For example: my target resolution is 1280x720, and the video feed I got is 1282x722, the 2px discrepancy caused blurriness.
If I manually adjust the window, I might get it to display pixel-perfectly again, and it is showing as 1280x720 size.

The screenshots I took are from a 2560x1440 monitor (height truncated to exclude window title and Windows 10 taskbar), it shows what it would look like when pixel-perfect AND when default resized (or with "." button). I attached a full-screen green color to help identify screen size given the video feed window generates an 1px edge already.

The issue is present with either MJPEG or H264 compression of the video feed, and when pixel-perfection is achieved, you could clearly tell the difference of blurriness and compression artefacts. The blurriness is not a compression artefact.
The issue is present with different browsers. I tried Chrome & Firefox.
The issue is NOT present when using the video feed full screen on a matching resolution display. e.g. showing 1920x1080 target feed full-screen on a 1080p display, or showing 1280x720 target feed full-screen on a 720p display. Meaning it's likely a window-scaling related issue that can be tuned in the webGUI software,
The issue is NOT present when using VNC client. Using TigerVNC, when display is set to 1:1 scaling, it shows pixel-perfect video feed even if it does not achieve full screen.

**Expected behavior**
Pixel-perfect video feed when default/resized, if viewing space allows.

**Screenshots**
1&2 720p feed manually adjusted to pixel-perfection [expected behavior]. full screen green display showing real feed size (you can count the pixel or parse it in image editing software)
[image]
[image]

3&4 Same as 1&2 BUT focus is lost
[image]
[image]

5&6 Default centered window size (or after hitting "." button to manually center&resize). Also showing full green screen for easy size comparison.
[image]
[image]

7&8 Same as 5&6 BUT focus is lost
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 21H1 (19043.1415)
 - Browser: Chrome_v97 (and Firefox)
 - VNC client: TigerVNC

**PiKVM info:**
  - Raspberry Pi board version: Pi 4 model B rev 1.4
  - PiKVM platform: v3 HAT (Kickstarter kit w/ steel case)
  - Video capture type:  CSI bridge
  - KVMD version: 3.52-1
  - uStreamer version: 4.11-1
  - Linux kernel: 5.10.52-14-ARCH

mdevaev, 2022-03-22:
Presumbly it's fixed in the new KVMD release 3.64. Could you confirm?

VoeGalore, 2022-03-22:
I can confirm one thing: when I focus/defocus the streaming window within the webGUI, the streaming window does not change size anymore, this is surely welcomed.

The extra pixel on window width is still present, please see attached screenshots (taskbar hidden, URL redacted):

default resizing with "." button
[image]

manually decrease width by **1px** (used to be 2px?)
[image]

VoeGalore, 2022-04-26:
Here are 2 additional screenshots taken with default stream window size (not pixel-perfect) and manually adjusted stream window size (pixel perfect). With the help of an additional screenshot tool you can see it's just 2 pixels wider than it needs to be to achieve pixel-perfect scaling.

[image]

[image]

VoeGalore, 2022-04-27:
Please check [https://github.com/pikvm/kvmd/pull/86](url) for potential fix of this issue, and [https://github.com/pikvm/kvmd/pull/87](url) for another minor fix of visual.

closed 2022-04-30 by mdevaev