pikvm/pikvm #604: Support USB Boot on official v3-HDMI-RPI4 image

cornytrace, 2022-01-11:
**Is your feature request related to a problem? Please describe.**
Booting from a fast USB3.0 USB stick or SSD results in a faster boot than using the microSD slot.

**Describe the solution you'd like**
Currently the v3-hdmi-rpi4 image hardcodes /dev/mmcblk0p2 as the root device. This results in an error to boot when flashing the image to a usb device.
by changing the cmdline.txt and /etc/fstab files to reference to UUID / PARTUUID instead of fixed device files this error can be avoided.

mdevaev, 2022-01-13:
Sup. I will schedule this in the next big kernel release.

arch1mede, 2022-01-21:
For anyone reading into this, this will be a major feature addition and will be announced on the Discord in #news once its available which may take months to implement as RPi devs are slow to fix kernel issues.

arch1mede, 2022-01-21:
> 

I would just like to add, you will be saving Micro Seconds, the work to put into making this use a ssd/usb is offset by the work saved. It already takes 20 seconds to boot and be able to access using a newer type class10 sd card.

Changing the default behavior would not be an official use case thereby making this more #self supported if anything were to happen to the functionality of pikvm.

mdevaev, 2022-03-21:
Hint for me:
- https://archlinuxarm.org/forum/viewtopic.php?f=67&t=14756
- https://forums.raspberrypi.com/viewtopic.php?t=280351
- https://thepi.io/how-to-boot-your-raspberry-pi-from-a-usb-mass-storage-device/

mdevaev, 2022-04-22:
We moved to partition labels. I think it should be enough from the PiKVM side. The rest of the tuning refers to Arch.

closed 2022-04-22 by mdevaev

fmnamado, 2023-12-19:
Hello,
I just installed pikvm.
Is it possible now to just clone the SD Card to SSD (with Clonezilla for example), or will it be impossible?
The idea is of course to boot from USB SSD.

Alternatively, if the above is true, is there a better way of installing pikvm or is it always flash to SD Card and then clone to SSD?

Feel free to share any advice regarding to this.

Thank you!

mdevaev, 2023-12-20:
[Please contact our support for any questions.](https://discord.gg/bpmXfz5)