pikvm/pikvm #608: Automatically create a iso/flash drive image from uploaded archive/zip file

rgl, 2022-01-15:
**Is your feature request related to a problem? Please describe.**

Firmware drivers (like the [Seeed Studio Odyssey X86J4105 board](https://wiki.seeedstudio.com/ODYSSEY-X86J4105-Installing-OS/#latest-bios-firmware)) are normally shipped as a zip file and in order to install them we have to use a physical flash drive or convert them to a flash drive image to use them with PiKVM. Both are a nuisance to do.

**Describe the solution you'd like**

The PiKVM Drive UI should automatically create a flash/iso image from an uploaded archive file (e.g. from a zip).

**Describe alternatives you've considered**

This can be manually done, but, since PiKVM already provides a pretty sweet UI, it should also support the above use case :-)

mdevaev, 2022-01-15:
Good idea

mdevaev, 2022-01-16:
Should we unzip file?

rgl, 2022-01-16:
Yes, it unzip the contents. Ideally, it would work like the imaging tools (e.g. Raspberry Pi Imager, Etcher), where they unzip, and detect the image type (e.g. if its an image inside the zip, it should just use it as-is).

And when I say (un)zip, it would ideally support other formats like gz (e.g. by using libarchive for handling the archives).