pikvm/pikvm #611: virtual keyboard layouts

GralfR, 2022-01-18:
My system:
PiKVM on RPi4 connected to a Win10 with USB-HID and HDMI-USB as third screen on intel integrated graphics
This Win10 is german, has a german keyboard and a german keyboard layout.

I'm connecting to the PiKVM from a Linux Mint 20.2 with Chrome 95.0.4638.54. The Linux system is also german with a german keyboard connected with a german layout.

Using the hardware keyboard from Linux via PiKVM into Windows 10 notepad is working as expected for language based characters and keys like äöüß or AltGr+q for @, AltGr+e for € etc.

Problem:
The virtual keyboard on PiKVM inside the browser has an english layout. I opened the virtual keyboard because shortcuts like Win+Shift+Left/right cursor do not work from the hardware-keyboard. This shortcut moves windows from one screen to another (piKVM is the third screen on Windows and mouse-mode in piKVM is on relative).

Request:
Make the virtual keyboard layout selectable to be able to match with hardware keyboard on the client. Perfect would be an automatic detection of the keyboard layout by detecting the hosts layout, but I don't think this is possible. As a workaround: match the virtual keyboard layout to the local/client layout.
A manual selection should always be possible.

maybe referred to #375 ?

mdevaev, 2022-01-20:
This is definitely related and will need to be done.

mdevaev, 2023-08-11:
This task has something in common with providing a native layout on Mac/PC (#1016). So I'm merging it with a newer issue, as it needs to be implemented all together.

wegethertravel, 2024-07-29:
Is there any luck in adding this feature?

mdevaev, 2024-07-29:
I hope to do this when we'll finish more priority tasks.

maxexcloo, 2024-08-23:
Definitely would use this, I keep mixing up my hotkeys 👍🏻

ykorzikowski, 2025-07-26:
I ran into the same problem like #1016. 

I solved it by using Microsoft Powertoys to archive a software side remapping. 

But I am looking for a hardware side solution. 

I was doing a bit research and found this: https://www.remapper.org/

It uses a Raspberry Pico and needs to be plugged in between the keyboard (or pikvm) and the host. 

I am not sure if it works but I will give it a try.

mdevaev, 2025-10-26:
In this task we also need to add AZERTY and Dvorak.