pikvm/pikvm #612: Share flash or drive-space

GralfR, 2022-01-18:
It's great to be able to upload a flash image and use that on the connected station as a flash-drive.
But I need a much faster way to be able to exchange files between host and slave i.e. remote and local station/webUI.

Some ideas:
1. A local connected flashdrive should be useable without the need to create and upload images. Perfect from a users perspective would be: plug a thumbdrive into the local station, select the drive inside the piKVM webUI for sharing and the piKVM emulates a flashdrive on the remote-station with access to the local thumbdrive.
There are some issues to deal with like: who has write-access. Perfect would be both stations. At least switchable inside the UI. This would help a lot with filetransfer from/to remote stations; e.g. download a driver file at the local station and copy it to a thumb drive, share this thumbdrive to the remote station and open/install the driver without the need to image a drive first.

2. The PiKVM should have a sharepoint for both ends. This sharepoint should be an emulated flashdrive on the remote-station and might be a webdav inside the webUI. Perfect would be write-access from both sides. That would help for topic 1 without a real thumbdrive.

I know, it's hard with two devices having write access to the same block-device. A workaround might be found in the [lineage OS](https://konstakang.com/devices/rpi4/). This is an Android on a Pi4.
I use this to emulate a MTP device at the USB-side of the pi and share the same folder via ethernet to another system. Thus having a sharepoint writeable from both systems without the need for switching write-access and both systems are not and must not be connected inside one single LAN.

Maybe something like that could be implemented into the PiKVM as well?

cipy, 2022-02-02:
1. 💯

mdevaev, 2022-07-26:
This will be a generic way to share files. It is possible to emulate virtual media without the need to use an image.