pikvm/pikvm #613: mouse/keyboard emulation not working on clean install of Raspbian OS buster/bullseye

cipy, 2022-01-18:
**Describe the bug**
A clear and concise description of what the bug is.

PiKVM doesn't seem to be able to control a Raspberry Pi computer

the video over HDMI is fine, but there's no interaction with the RPi OS

**To Reproduce**
Steps to reproduce the behavior:
1. install a fresh Raspbian OS on a RPi 4B
2. connect it to a working PiKVM HAT v3
3. try to interact using the mouse/keyboard
4. oops/fail

**Expected behavior**
Should work fine.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Raspbian OS Buster
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version RPi 4B
  - PiKVM platform v3
  - Video capture type CSI bridge
  - latest `pacman -Syu`
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

Thanks,

arch1mede, 2022-01-18:
Hi, can you join discord so we can troubleshoot this, it seems like there is maybe an issue with the v3 usbc bridge or cable. I would rather not do troubleshooting in this space.

mdevaev, 2022-01-21:
This is the same problem as #614

closed 2022-01-21 by mdevaev